google-deepmind/mujoco_menagerie #1: How to attach 2f85 hand to ur5e arm ?

jrabary, 2022-09-12:
I would like to attach the robotiq2f85 hand to ur5e arm. I create an attachement site under the wrist3 of the ur5e and attached the hand to it. I use pymjcf from dm_control to create the model file as follow:

```python
from dm_control import mjcf

scene = mjcf.RootElement()
scene.asset.add('texture', type="2d", name="groundplane", builtin="checker", mark="edge", rgb1="0.2 0.3 0.4", rgb2="0.1 0.2 0.3", markrgb="0.8 0.8 0.8", width="300", height="300")
scene.asset.add('material', name="groundplane", texture="groundplane", texuniform="true", texrepeat="5 5", reflectance="0.2")

scene.worldbody.add('light', pos="0 0 1")
scene.worldbody.add('light', pos="0 -0.2 1", dir="0 0.2 -0.8", directional="true")
scene.worldbody.add('geom', name='floor', size="0 0 0.05", type='plane', material='groundplane')

arm = mjcf.from_path('universal_robots_ur5e/ur5e.xml')
wrist = arm.find('body', 'wrist_3_link')
attachement_site = wrist.add('site')

hand = mjcf.from_path('robotiq_2f85/2f85.xml')
attachement_site.attach(hand)

scene.attach(arm)

mjcf.export_with_assets(scene, "scene_2f85", "scene_test.xml")
```

the file `scene_text.xml` is generated correctly the hand is not visible when loading the file into the simulator.

kevinzakka, 2022-09-12:
It's probably under the plane and you can't see it. Make sure you are attaching the gripper with a position / orientation offset. Try something like:

```python
frame = attachement_site.attach(hand)
frame.quat = "-1 1 0 0"
frame.pos = "0 0.1 0"
```

[image]

saran-t, 2022-09-12:
The idiomatic way to do this in PyMJCF is:

```
wrist = arm.find('body', 'wrist_3_link')
attachment_site = wrist.add('site', pos=[0, 0.1, 0], quat=[-1, 1, 0, 0])
attachment_site.attach(hand)
```

closed 2022-09-12 by kevinzakka

jrabary, 2022-09-13:
Thanks it's working but need to delete the keyframe definition from `ur5e.xml`. 

```
  <keyframe>
    <key name="ur5e/home" qpos="-1.5708 -1.5708 1.5708 -1.5708 -1.5708 0" ctrl="-1.5708 -1.5708 1.5708 -1.5708 -1.5708 0"/>
  </keyframe>
```

As far as I understand it's for initialization and don't have any effect on the simulation. But, I'm curious to know how should one update this field to take into account the attached hand ?

kevinzakka, 2022-09-13:
You'll have to increase the dimensionality of the qpos to account for the dofs of the gripper. If you load the corresponding saved XML file, it should tell you the length it expects btw.

kevinzakka, 2022-10-21:
Hi @jrabary, I added some [helper code](https://github.com/deepmind/mujoco_menagerie/blob/main/FAQ.md#how-do-i-attach-a-hand-to-an-arm) for attaching Menagerie end-effectors to Menagerie arms.

joaopfg, 2022-10-31:
@kevinzakka @saran-t If I add the gripper body tree directly as a child to the wrist_3_link body tree in the XML, will I still need the site ? 

It seems to work well in my simulation. I'm asking to know if I'm missing something.

zxwang96, 2023-04-18:
> It's probably under the plane and you can't see it. Make sure you are attaching the gripper with a position / orientation offset. Try something like:
> 
> ```python
> frame = attachement_site.attach(hand)
> frame.quat = "-1 1 0 0"
> frame.pos = "0 0.1 0"
> ```
> 
> [image]

Hi, I was wondering if you have a more precise position offset value? I tried pos = "0 0.1 0" but found the sim visualization is different from the real-world setup. Thanks!