google-deepmind/mujoco_menagerie #4: Add a separate hand model for Franka Panda

jrabary, 2022-10-14:
Currently the Franka arm is available in a a full and no hand model. It would be useful to have the hand only model. 
Especially, it will be easier to use it with dm_control manipulation task that expects a separate arm and hand model.

jrabary, 2022-10-14:
I'm not sure if I get it right but in order to make it work I needed to change the attachment site in `panda_nohand` to 
```xml
<site name="attachment_site" pos="0 0 0.107" quat="0.9238795 0 0 -0.3826834"/>
```
and the `hand` definition be like

```xml
<mujoco model="panda_link8">
  <compiler angle="radian" meshdir="assets" autolimits="true"/>

  <option integrator="implicit"/>

  <default>
    <default class="panda">
      <material specular="0.5" shininess="0.25"/>
      <joint armature="0.1" damping="1" axis="0 0 1" range="-2.8973 2.8973"/>
      <general dyntype="none" biastype="affine" ctrlrange="-2.8973 2.8973" forcerange="-87 87"/>
      <default class="finger">
        <joint axis="0 1 0" type="slide" range="0 0.04"/>
      </default>

      <default class="visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2"/>
      </default>
      <default class="collision">
        <geom type="mesh" group="3"/>
        <default class="fingertip_pad_collision_1">
          <geom type="box" size="0.0085 0.004 0.0085" pos="0 0.0055 0.0445"/>
        </default>
        <default class="fingertip_pad_collision_2">
          <geom type="box" size="0.003 0.002 0.003" pos="0.0055 0.002 0.05"/>
        </default>
        <default class="fingertip_pad_collision_3">
          <geom type="box" size="0.003 0.002 0.003" pos="-0.0055 0.002 0.05"/>
        </default>
        <default class="fingertip_pad_collision_4">
          <geom type="box" size="0.003 0.002 0.0035" pos="0.0055 0.002 0.0395"/>
        </default>
        <default class="fingertip_pad_collision_5">
          <geom type="box" size="0.003 0.002 0.0035" pos="-0.0055 0.002 0.0395"/>
        </default>
      </default>
    </default>
  </default>

  <asset>
    <material class="panda" name="white" rgba="1 1 1 1"/>
    <material class="panda" name="off_white" rgba="0.901961 0.921569 0.929412 1"/>
    <material class="panda" name="black" rgba="0.25 0.25 0.25 1"/>
    <material class="panda" name="green" rgba="0 1 0 1"/>
    <material class="panda" name="light_blue" rgba="0.039216 0.541176 0.780392 1"/>

    <!-- Collision meshes -->
    <mesh name="hand_c" file="hand.stl"/>

    <!-- Visual meshes -->
    <mesh file="hand_0.obj"/>
    <mesh file="hand_1.obj"/>
    <mesh file="hand_2.obj"/>
    <mesh file="hand_3.obj"/>
    <mesh file="hand_4.obj"/>
    <mesh file="finger_0.obj"/>
    <mesh file="finger_1.obj"/>
  </asset>

  <worldbody>
    <body name="hand">
      <inertial mass="0.73" pos="-0.01 0 0.03"
          diaginertia="0.001 0.0025 0.0017"/>
      <geom mesh="hand_0" material="off_white" class="visual"/>
      <geom mesh="hand_1" material="black" class="visual"/>
      <geom mesh="hand_2" material="black" class="visual"/>
      <geom mesh="hand_3" material="white" class="visual"/>
      <geom mesh="hand_4" material="off_white" class="visual"/>
      <geom mesh="hand_c" class="collision"/>
      <body name="left_finger" pos="0 0 0.0584">
        <inertial mass="0.015" pos="0 0 0"
            diaginertia="2.375e-6 2.375e-6 7.5e-7"/>
        <joint name="finger_joint1" class="finger"/>
        <geom mesh="finger_0" material="off_white" class="visual"/>
        <geom mesh="finger_1" material="black" class="visual"/>
        <geom mesh="finger_0" class="collision"/>
        <geom class="fingertip_pad_collision_1"/>
        <geom class="fingertip_pad_collision_2"/>
        <geom class="fingertip_pad_collision_3"/>
        <geom class="fingertip_pad_collision_4"/>
        <geom class="fingertip_pad_collision_5"/>
      </body>
      <body name="right_finger" pos="0 0 0.0584" quat="0 0 0 1">
        <inertial mass="0.015" pos="0 0 0"
            diaginertia="2.375e-6 2.375e-6 7.5e-7"/>
        <joint name="finger_joint2" class="finger"/>
        <geom mesh="finger_0" material="off_white" class="visual"/>
        <geom mesh="finger_1" material="black" class="visual"/>
        <geom mesh="finger_0" class="collision"/>
        <geom class="fingertip_pad_collision_1"/>
        <geom class="fingertip_pad_collision_2"/>
        <geom class="fingertip_pad_collision_3"/>
        <geom class="fingertip_pad_collision_4"/>
        <geom class="fingertip_pad_collision_5"/>
      </body>
    </body>
  </worldbody>

  <!--contact>
    <exclude body1="link7" body2="link8"/>
  </contact -->

  <tendon>
    <fixed name="split">
      <joint joint="finger_joint1" coef="0.5"/>
      <joint joint="finger_joint2" coef="0.5"/>
    </fixed>
  </tendon>

  <equality>
    <joint joint1="finger_joint1" joint2="finger_joint2" solimp="0.95 0.99 0.001" solref="0.005 1"/>
  </equality>

  <actuator>
    <!-- Remap original ctrlrange (0, 0.04) to (0, 255): 0.04 * 100 / 255 = 0.01568627451 -->
    <general class="panda" name="actuator8" tendon="split" forcerange="-100 100" ctrlrange="0 255"
        gainprm="0.01568627451 0 0" biasprm="0 -100 -10"/>
  </actuator>

  <keyframe>
    <key name="home" qpos="0.04 0.04"
        ctrl="255"/>
  </keyframe>
</mujoco>
```

kevinzakka, 2022-10-15:
Thanks for the request, I'll submit a PR to add a hand only model!

commit b4d7d138 (2022-10-17): Add hand only panda model.

Also remove unnecessary `link8` from `panda.xml`.

Fixes #4.

PiperOrigin-RevId: 481714182
Change-Id: Ie9b46921d2a58f828d0f5e2d6a97032a6c018eb0
closed 2022-10-17 by copybara-service[bot]