google-deepmind/mujoco_menagerie #5: Stepping simulation changes robot configuration

rpapallas, 2022-10-18:
**Which model is the issue affecting?** Franka Emika Panda

**What is the issue?** I load the scene in Python (using the `mujoco.MjModel.from_xml_path` function) and then change the arm configuration (set `qpos`) to some default/home position (see figure 1). Then if I step the simulation (mujoco.mj_step()) the robot configuration changes instantaneously to figure 2. I can't seem to move the arm either; sending controls to actuators using the following code don't seem to do anything:

```
        for i in range(self.joint_names):
            qfrc = self.data.joint(joint_name).qfrc_bias
            self.data.actuator(f'actuator{i+1}').ctrl = qfrc + controls[i]
```

[image]

[image]

Is this a problem with the model or something wrong on my side? Did you try actuating the specific model?

rpapallas, 2022-10-19:
I see; thank you for your response @kevinzakka. I didn't realize that the `<general>` actuator is a position servo!

closed 2022-10-19 by rpapallas

kevinzakka, 2022-10-19:
@rpapallas yeah I'd recommend you open the model in the simulate binary and then play with the actuators in the right pane!