google-deepmind/mujoco_menagerie #7: Force/Torque Sensor!

KhalilAbuibaid, 2022-11-10:
Hey there, as it is known, ur5e has FT sensor at its six axis. I added FT sensor by using body/ site of the sixth axis, sensor/ force and sensor/ torque of that site. Then, when I mount the robotiq's gripper, it takes the gripper force (i.e. weight) into consideration. Is there any way I can overcome this weight and calibrate the sensor to start zero?

HoangGiang93, 2022-11-11:
I think you should add FT sensor on the gripper, not on the sixth axis. If you add the FT to the sixth axis, of course it will measure the weight of the gripper also.

yuvaltassa, 2022-11-11:
As @HoangGiang93 says, the sensor will always measure whatever comes after it, in MuJoCo and in the real world. I don't see a way to calibrate that away, as the force/torque is pose-dependent.

closed 2022-11-11 by yuvaltassa