google-deepmind/mujoco_menagerie #10: Trying to create a UR10 model -- visual orientations are wrong

alper111, 2023-02-05:
Hi. I am trying to create a UR10e model by following the steps in https://github.com/deepmind/mujoco_menagerie/blob/main/universal_robots_ur5e/README.md. However, when I load the model, visuals are not in the correct orientation. Below is a screenshot of the model:

[image]
[image]

I thought maybe I have missed a step while converting from DAE to OBJ in Blender, and I realized that the rotation is not [0, 0, 0] when I imported DAE files (but [90, 0, -180]). So I have changed them to [0, 0, 0] and re-converted the files. Visuals have changed, but they are still incorrect:

[image]
[image]

I think the only difference between ur5e and ur10e should be their sizes and lengths, not orientations. I uploaded my changes to a repo if you want to check it: https://github.com/alper111/ur10e_mujoco. There may be some other issues such as colors/Kp values and so on, but they are not relevant at the moment.

Blender: Version 3.4.1 (2022-12-20)
Mujoco: Version 2.3.1
System: macOS Ventura 13.1
Architecture: arm

kevinzakka, 2023-02-14:
Hi @alper111, I assume you have DAE or STL meshes and you're converting them to OBJ? I used Blender for this process and had to make sure I was saving the meshes with Z point up. Can you double check if you're doing that?

alper111, 2023-02-16:
Hi Kevin. Pointing them upwards Z worked. I reinitialized the https://github.com/alper111/ur10e_mujoco repo with all the new changes. I mostly followed your conventions, but I am not opening a PR because I didn't do any system identification. Feel free to copy it for future releases, though.

closed 2023-02-16 by alper111

kevinzakka, 2023-02-17:
@alper111 Awesome! Feel free to submit a PR, sys ID is planned for later this year anyway.