google-deepmind/mujoco_menagerie #11: Is it possible to provide example code for usage?

elevenjiang1, 2023-02-09:
Hello, great models provide which can help a freshman to use Mujoco !

And I am curious about that is it possible to provide the code to implement the videos shown in [YouTube ](https://www.youtube.com/watch?v=wi_zJzRm8Ic)？

This will help me a lot to use Mujoco better!

Thank you~

kevinzakka, 2023-02-14:
Hi @elevenjiang1, I'll try to push out a Colab notebook with code for generating the teaser videos with the next release (scheduled for April).

elevenjiang1, 2023-02-14:
Thanks a lot !~

kevinzakka, 2023-02-14:
Let's keep this open until we release the Colab!

joe-louca, 2023-05-05:
Hi @kevinzakka, These look like really great models! Any update on the when a Colab might be released?

kevinzakka, 2023-05-05:
Hi @joe-louca, do you want the colab for the purpose of rendering a blend of the visual and collision meshes or a Colab tutorial for how to use these models?

joe-louca, 2023-05-09:
@kevinzakka - a tutorial for how to use the models would be great please. I'd like to recreate something like the demo videos as a starting point so mainly looking to do some joint control and compute contact forces

kevinzakka, 2023-05-09:
@joe-louca V2 of menagerie will be out super soon (this month), and I'll add the Colab tutorial that generated the YouTube videos!

joe-louca, 2023-05-09:
Oh perfect. That's great, thanks a lot!

jlhalv92, 2023-06-13:
Dear @kevinzakka, any news about the release date for Menagerie V2?

Thanks a lot for the update :)

kevinzakka, 2023-06-13:
@jlhalv92 We released a bunch of models last month. There should be 1 or 2 more coming out soon, but you could consider those V2 already.

joe-louca, 2023-06-13:
Thanks for the update @kevinzakka. Just checked out some of the new models, that's a massive help!

xiongjunyan0, 2023-06-27:
Hi @kevinzakka  , may i ask about  any news or update on how to implement these model on colab?

NilsDengler, 2023-08-22:
Are you still planning to release the Colab at some point?

kevinzakka, 2023-09-25:
I'll have a Colab notebook out by the end of this week. You'll be able to load a model and simulate it with frame blending of visual and collision meshes.

In terms of how to use these models, there's nothing special to it. It's like any other MuJoCo model. I'd recommend you go through the 2 MuJoCo tutorials ([tutorial 1](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/python/tutorial.ipynb), [tutorial 2](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/python/LQR.ipynb)) and then take a look at [dm_control](https://github.com/google-deepmind/dm_control), [dm_robotics](https://github.com/google-deepmind/dm_robotics), and [mujoco_mpc](https://github.com/google-deepmind/mujoco_mpc) to see how these models can be used to do RL or optimal control.

closed 2023-10-02 by copybara-service[bot]

kevinzakka, 2023-10-02:
Super minimal [tutorial notebook](https://github.com/google-deepmind/mujoco_menagerie/blob/main/tutorial.ipynb) is now live.

commit fb2ae3bc (2023-10-02): Open-source notebook for simulating and rendering Menagerie models. Fixes #11.

PiperOrigin-RevId: 570130413
Change-Id: Iaa1f88cf4cfc7044a56a81384679ccc91ae93bd7