google-deepmind/mujoco_menagerie #12: Contributing to system identifications suggestions

raghavauppuluri13, 2023-02-21:
Hello! Thanks for taking the initiative to consolidate a set of high-quality models for mujoco! In our research lab, we have the physical models for the panda, ur5e, robotiq gripper. Are there any recommended experiments for system ID that we could do to contribute to improving the current quality of these models?

yuvaltassa, 2023-02-22:
We are planning a sysID package which we will release publicly and also use for the robots that we have hardware access to. The package documentation will contain information on data collection best practices.

I will update here when the package is released or when we have updated the robot models you mentioned.

Thanks!

closed 2023-02-22 by yuvaltassa

peterdavidfagan, 2023-11-29:
+1 on this sysID package this would be great to have access to even if only for learning/teaching purposes.