google-deepmind/mujoco_menagerie #13: How to add multiple instances of the same robot?

cemoke, 2023-02-25:
Hi, 
I would like to have a bimanual setup consisting of two franka emika panda robots. What do you think the best way is to do this without getting a repeated names error? I think I should add a prefix or a namespace but I don't know how I can do it easily. I would be glad if you could guide me.
Thanks in advance.

kevinzakka, 2023-02-25:
Are you using dm_control or the native python bindings?

cemoke, 2023-02-25:
Native python bindings

kevinzakka, 2023-02-25:
The include mechanism for MuJoCo is pretty limiting at the moment. I think your best bet is to create a new XML file with two repeated panda kinematics trees in the worldbody. You should be able to just have one global definition of everything else (assets, meshes, etc.) Make sure you give the bodies, joints and actuators a unique name (I would just prepend a `panda1/` and `panda2/` for each robot definition).

kevinzakka, 2023-02-27:
Closing this for now. Please re-open if you run into any issues.

closed 2023-02-27 by kevinzakka

guangnianyuji, 2024-05-29:
> Hi, I would like to have a bimanual setup consisting of two franka emika panda robots. What do you think the best way is to do this without getting a repeated names error? I think I should add a [...]

Hello! Do you do it successfully?I do make sure myself give the bodies, joints and actuators a unique name, but the manipulation is in a mess, they look like floating. Can you show me your xml script ?