google-deepmind/mujoco_menagerie #15: Finger tip positions

Masoudghanei, 2023-04-05:
Hello, for E3M5 hand model, How can I extract the cartesian positions of finger tips?
xanchor gives me the joints positions but, I also need the fingertips positions.

Masoudghanei, 2023-04-06:
I could handle it by adding site to the finger tips in xml file.

kevinzakka, 2023-04-06:
@Masoudghanei Adding sites and querying their `xpos` is indeed the mujoco thing to do!

Masoudghanei, 2023-04-06:
@kevinzakka thank you for your answer, I have another problem.
I want to save positions related to each frame but, it just saves the positions of final frame.
I use this code:

positions = []
frames = [] 
mujoco.mj_step(model, data) 
if len(frames) < data.time * framerate:
        renderer.update_scene(data)
        pixels = renderer.render().copy()
        frames.append(pixels)
        positions.append(data.site_xpos)
        

media.show_video(frames, fps=framerate)

yuvaltassa, 2023-04-06:
data.site_xpos.copy()

closed 2023-04-06 by yuvaltassa