google-deepmind/mujoco_menagerie #16: Version compatibility issue

tadashiK, 2023-05-14:
**Which model is the issue affecting?**

UR5e and Panda.

**What is the issue?**

In UR5e and Panda, implicitfast integrator is used. However, it is a new feature of MuJoCo introduced in version 2.3.3, which is much newer than the prerequisite version (2.2.2). So, README needs to be updated to reflect this change of the integrator.

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

kevinzakka, 2023-05-15:
Hi @tadashiK, we should definitely update the README for each robot to specify the minimum MuJoCo version. Note also that we provide [tagged releases](https://github.com/deepmind/mujoco_menagerie/tags), so you should be able to use a previous version that is compatible with 2.2.2.

tadashiK, 2023-05-16:
Hi @kevinzakka, thanks a lot for the reply. Yes, I can use an old version compatible with MuJoCo 2.2.2. I just wanted to let you know that the current README is outdated :smiley:

kevinzakka, 2023-05-16:
@akfidjeland Noted and appreciated, I'll make a push soon!

closed 2023-05-16 by copybara-service[bot]