google-deepmind/mujoco_menagerie #19: Add TIAGo and TIAGo++ Robot from PAL Robotics

saikishor, 2023-07-26:
**URL for the existing model source (e.g. URDF)**

TIAGo URDF : https://github.com/pal-robotics/tiago_robot/tree/kinetic-devel/tiago_description/robots
TIAGo++ URDF: https://github.com/pal-robotics/tiago_dual_robot/tree/kinetic-devel/tiago_dual_description/robots

**License terms of the upstream model**

Both TIAGo and TIAGo++ will be with [Apache License 2.0](https://github.com/pal-robotics/tiago_robot/blob/humble-devel/tiago_description/package.xml#L12)

**What do you think is required to convert and tune this model for MuJoCo?**
* Being able to convert from the URDF to MuJoCo using the `compile`  command
* We need to add the actuators to the model for better joint actuation
* Stability of the model w.r.t to the actuator's gain tuning
* Taking into account the existing issue with the repeated naming in the [MuJoCo](https://github.com/deepmind/mujoco/issues/163)

**Are you proposing to help develop and maintain the new model?**

Yes, we can develop and help maintain the new model. It is in our interest as a part of the @pal-robotics team.

saikishor, 2023-08-04:
Hello!

Can we open a PR on this from our end?

Thank you,

Best Regards,
Sai

kevinzakka, 2023-08-09:
Cool, thanks for making the PR. Working on integrating on my end!

saikishor, 2023-08-10:
Hello @kevinzakka!

Thank you for your imminent response. Looking forward to seeing them in the main branch soon. If you have any queries or need any changes, please feel free to get back to us.

Thank you,

Best Regards,
Sai

kevinzakka, 2023-08-24:
Hi @saikishor, noticing weird behavior with the robot fingers. Have those been carefully looked at?

saikishor, 2023-08-24:
Hello @kevinzakka!

I just checked it. Unfortunately, we missed that detail. The mistake is on our end. I'm sorry about that. 

https://github.com/pal-robotics-forks/mujoco_menagerie/blob/140ae8d30b430d9d8d8f0c42e031b93b59cb2968/pal_tiago/tiago.xml#L239-L245

I think the gains for those joints need to be tweaked. Tomorrow, we will try to fix that. Thank you for reporting it. 

Do you have any other feedback?

Thank you,

Best Regards,
Sai Kishor Kothakota

saikishor, 2023-08-25:
Hey @kevinzakka!

Thank you for reporting the issues. We have pushed some new changes to the PR. Whenever you get some time, please kindly try it and let us know. 

Thank you,

Best Regards,
Sai

kevinzakka, 2023-08-31:
Thanks @saikishor! I'm wrapping up another robot submission. Once that's done, I can get your PR reviewed and merged in. Thanks for being patient.

saikishor, 2023-08-31:
Thank you so much @kevinzakka.

PR #94 (2024-11-22): Add TIAGo and TIAGo++ Robots from PAL Robotics (updated)
This PR adds the requested models as in the issue: https://github.com/google-deepmind/mujoco_menagerie/issues/19. The models are generated inspiring from the existing models in the repository.

Please feel free to suggest any suggestions to improve the quality of this PR.

We have applied the same suggestions, @kevinzakka and @yuvaltassa have given us on #22 for the PR of #20 to ease the process. 

We hope this is helpful.

closes: #19 
closes: #20
closed 2024-11-22 by copybara-service[bot]