google-deepmind/mujoco_menagerie #28: Add WidowX

vyeevani, 2023-10-11:
If you would like to see a new model added to the Menagerie collection, please provide us with the following information _before_ sending a pull request. (Ideally, before you start working on converting or tuning the model for MuJoCo.)

Given recent work from Google like the Open X Embodiment dataset, it would be cool to try policies entirely in simulation. Given my personal workflow is primarily in Brax as opposed to the Gazebo (the native deployment of interbotix), it would be cool to have this integrated.

**URL for the existing model source (e.g. URDF)**
https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_xsarms/interbotix_xsarm_descriptions/urdf

**License terms of the upstream model**
BSD 3-Clause License

**What do you think is required to convert and tune this model for MuJoCo?**
You'd have to run the model through argument substitution using ROS's xacro module.

**Are you proposing to help develop and maintain the new model?**
Sure

kevinzakka, 2023-12-05:
Hey @vyeevani, thanks for your interest in Menagerie. The WidowX and ViperX will be out this week hopefully. Cheers!

kevinzakka, 2023-12-05:
The ViperX 300 6DOF is now live as of 477ec56566682d663b996d86a188c75499e74b98.

ygtkorkmaz, 2024-04-15:
Is there any update on WidowX?

kevinzakka, 2024-06-04:
@ygtkorkmaz Will be out by end of week.

commit 3ff28577 (2024-06-04): Add WidowX 250 6-DoF. Closes #28.

PiperOrigin-RevId: 640239380
Change-Id: I486284c01ff41957043ec5f6cae907d6489ff52c
closed 2024-06-04 by copybara-service[bot]