google-deepmind/mujoco_menagerie #29: Add the new model of Unitree Go2

RayYoh, 2023-10-21:
If you would like to see a new model added to the Menagerie collection, please provide us with the following information _before_ sending a pull request. (Ideally, before you start working on converting or tuning the model for MuJoCo.)

**URL for the existing model source (e.g. URDF)**: [https://github.com/unitreerobotics/unitree_ros/tree/master/robots](Unitree Go2)

**License terms of the upstream model**: BSD 3-Clause License

**What do you think is required to convert and tune this model for MuJoCo?**: Go2 is a new version of Go1 that has been imrpoved.

**Are you proposing to help develop and maintain the new model?**

kevinzakka, 2023-10-22:
Hey @RayYoh, it will be released within the week.

closed 2023-10-24 by copybara-service[bot]

kevinzakka, 2023-10-24:
@RayYoh Model is live.

commit 2ffe3e33 (2023-10-24): Add Unitree Go2 quadruped. Resolves #29.

PiperOrigin-RevId: 576172175
Change-Id: I201858fff037d064812ea202885487aea8648c63