google-deepmind/mujoco_menagerie #30: Add UFactory Lite6

peterdavidfagan, 2023-10-30:
**URL for the existing model source (e.g. URDF)**
[Lite6 URDF](https://github.com/xArm-Developer/xarm_ros2/blob/master/xarm_description/urdf/lite6/lite6.urdf.xacro
)
**License terms of the upstream model**
[BSD 3-Clause](https://github.com/xArm-Developer/xarm_ros2/blob/master/LICENSE
)
**What do you think is required to convert and tune this model for MuJoCo?**

- Being able to convert from the URDF to MuJoCo using the compile command
- Tuning model parameters for actuators

**Are you proposing to help develop and maintain the new model?**
I would be happy to help develop and maintain the new model.

kevinzakka, 2023-10-31:
Hi @peterdavidfagan, if you can get me the URDF files (not xacro), I can have this out pretty fast.

peterdavidfagan, 2023-11-01:
Thanks @kevinzakka, I'll clean the URDF  I have (remove ros 2 control tags etc.) and link it here shortly.

peterdavidfagan, 2023-11-01:
Please find files at the following [link](https://github.com/peterdavidfagan/lite6_urdf_2_mjcf/tree/main/urdf). This loads into the mjviewer so I believe all that is needed is there, if something is missing I'd be happy to add it. Also I haven't included the grippers here but I'd be happy to add these (I am currently working with a custom gripper that someone else open sourced.)

[image]

kevinzakka, 2023-11-02:
Hi @peterdavidfagan, did it have any DAE or OBJ files? It would be nice to have pretty visual meshes.

kevinzakka, 2023-11-02:
If it doesn't exist in the original description, then we can just add textures to the STLs!

kevinzakka, 2023-11-02:
@peterdavidfagan Model should be released by tomorrow at the latest, cheers :)

peterdavidfagan, 2023-11-02:
> If it doesn't exist in the original description, then we can just add textures to the STLs!

This is great to know I wasn't aware of this, I have less experience working with 3D modelling (although I have been exploring Blender a bit recently). 

> @peterdavidfagan Model should be released by tomorrow at the latest, cheers :)

This is awesome, thank you so much @kevinzakka.

closed 2023-11-03 by copybara-service[bot]

peterdavidfagan, 2023-11-04:
Thank you @kevinzakka.

commit cbbc575b (2023-11-03): Add UFACTORY Lite 6 robot arm. Fixes #30.

PiperOrigin-RevId: 579222088
Change-Id: Ief06b2b5ca38939ece0eb77735f119c41cc2d067