google-deepmind/mujoco_menagerie #33: Franka Arm with Robotiq Gripper

pushkalkatara, 2023-11-28:
Hi, Thanks for open-sourcing all robot models.

I wanted to use the [Robotiq Gripper](https://github.com/google-deepmind/mujoco_menagerie/blob/main/robotiq_2f85/2f85.xml) with [Franka Arm](https://github.com/google-deepmind/mujoco_menagerie/blob/main/franka_emika_panda/panda_nohand.xml) and tried integrating the xmls together without luck.

Is there any documentation to merge two robot models? Thanks!

kevinzakka, 2023-11-28:
Are you using native MuJoCo bindings? If so, you need to namescope the default classes. That means prefixing the panda default classes with "panda/" and robotiq ones with "2f85/". Otherwise, if you're using dm_control's MJCF, see the [FAQ](https://github.com/google-deepmind/mujoco_menagerie/blob/main/FAQ.md#how-do-i-attach-a-hand-to-an-arm).

kevinzakka, 2023-11-29:
Marking as resolved, feel free to re-open if you get stuck.

closed 2023-11-29 by kevinzakka