google-deepmind/mujoco_menagerie #37: Attaching the RobotIQ 2f85 gripper to the UR10e robotic arm.

ChristosPeridis, 2023-12-19:
**With 2f85 gripper**

**When combing the tow xml files for the UR10e and the 2f85 gripper, the 2f85 gripper appears to be glitching in the simulation.**

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

Hello dear DeepMind engineers,

I hope you and everyone in your family are all well and healthy!!! I am a Research Assistant in Lifelong Deep Reinforcement Learning in Loughborough University. I am a member of the University's [Intelligent Automation Lab](https://www.lboro.ac.uk/research/intelligent-automation/) which focuses a lot on developing intelligent solutions for industrial robotics scenarios. I am working on the Industrial Robots as a Service (IRaaS) project, where I am tasked to solve industrial robotics tasks, such as pick and place, using Deep Reinforcement Learning techniques. For that reason I have worked using the MuJoCo simulated environments to test the DRL algorithms we are developing in our lab, before moving to an actual physical robot. My supervisor requested me to create a Pick and Place environment with the UR10e robot, which our lab is equipped with, fitted with the RobotIQ 2f85 gripper as an end-effector. I have performed this works by adding to the same directory the .obj files for the UR10e and the .stl files for the 2f85. Then, I combined the two xml files for the UR10e and the 2f85 to a new .xml called 'ur10e_2f85.xml'. In this file I have separated the default values and the actuators as much as possible to avoid any conflicts. The 2f85 has been attached to the 'wrist_3_link' and its position matches the position of the 'attachment_site'. Further more I have added contact exclusion of the different parts of the 2f85 to the  'wrist_3_link' of the UR10e to make sure that unwanted collision might cause this behavior. This slightly improved the gripper's behavior but did not fixed it completely. Furthermore, I have experimented with the '<option integrator="implicitfast"/> and the <option impratio="10"/> parameters to see how they might affect the integration of the two parts. I observed that using both option parameters made the 2f85 gripper to completely collapse after a few MuJoCo simulation steps. Using only the impartion option with the value of 10 seemed to be the most stable option. I will provide you with the ur10e_2f85.xml file that I have created, the Python script that I have developed and been using for performing the opening and closing of the gripper. Furthermore, I am providing you a video clip of the script running with the scene.xml for the UR10e_2f85 combo and one videoclip with the same script running with the scene.xml for the 2f85 only, to demonstrate the glitches occurring in the combo and the normal execution that it is achieved by simply running the script with just the gripper model. The video with the combo is demonstrating the combo's behavior after all the adjustments I have made in my attempt to resolve the issue, as described above. All code files, along with the appropriate scene.xml files for each of the two cases, are included in the supplied .zip file.
[UR10e_2f85_xml_plus_Python_manipulation_script.zip](https://github.com/google-deepmind/mujoco_menagerie/files/13715269/UR10e_2f85_xml_plus_Python_manipulation_script.zip)

https://github.com/google-deepmind/mujoco_menagerie/assets/34765945/f9e89c95-544f-4763-9dab-7b1a003688d0

https://github.com/google-deepmind/mujoco_menagerie/assets/34765945/469e97e7-c4a1-48d8-ae2e-e2def75a5233

Can you please help me detect the cause of this issue and resolve it?

Thank you very much in advance for the valuable help and support!!!

Kind regards,

Christos Peridis

kevinzakka, 2023-12-19:
Here's an example XML merging the UR5e and the Robotiq gripper:

<details><summary>Toggle to see XML</summary>

```xml
<mujoco model="ur5e">
  <compiler angle="radian" autolimits="true"/>

  <option integrator="implicitfast"/>

  <default>
    <default class="ur5e">
      <default class="visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2"/>
      </default>
      <default class="collision">
        <geom type="capsule" group="3"/>
        <default class="eef_collision">
          <geom type="cylinder"/>
        </default>
      </default>
      <joint axis="0 1 0" range="-6.28319 6.28319"/>
      <default class="joint_limited">
        <joint range="-3.1415 3.1415"/>
      </default>
      <default class="position">
        <position ctrlrange="-6.2831 6.2831" kp="2000" kv="200" forcerange="-150 150"/>
        <default class="position_limited">
          <position ctrlrange="-3.1415 3.1415"/>
        </default>
        <default class="position_small">
          <position kp="500" kv="50" forcerange="-28 28"/>
        </default>
      </default>
    </default>
    <default class="2f85">
      <mesh scale="0.001 0.001 0.001"/>
      <general biastype="affine"/>
      <site type="sphere" rgba="0.9 0.9 0.9 1" size="0.005" group="4"/>
      <joint axis="1 0 0"/>
      <default class="driver">
        <joint range="0 0.8" armature="0.005" damping="0.1" solimplimit="0.95 0.99 0.001" solreflimit="0.005 1"/>
      </default>
      <default class="follower">
        <joint range="-0.872664 0.872664" pos="0 -0.018 0.0065" solimplimit="0.95 0.99 0.001" solreflimit="0.005 1"/>
      </default>
      <default class="spring_link">
        <joint range="-0.29670597283 0.8" stiffness="0.05" springref="2.62" damping="0.00125"/>
      </default>
      <default class="coupler">
        <joint range="-1.57 0" solimplimit="0.95 0.99 0.001" solreflimit="0.005 1"/>
      </default>
      <default class="2f85/visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2"/>
      </default>
      <default class="2f85/collision">
        <geom type="mesh" group="3"/>
        <default class="pad_box1">
          <geom mass="0" type="box" pos="0 -0.0026 0.028125" size="0.011 0.004 0.009375" friction="0.7"
            solimp="0.95 0.99 0.001" solref="0.004 1" priority="1" rgba="0.55 0.55 0.55 1"/>
        </default>
        <default class="pad_box2">
          <geom mass="0" type="box" pos="0 -0.0026 0.009375" size="0.011 0.004 0.009375" friction="0.6"
            solimp="0.95 0.99 0.001" solref="0.004 1" priority="1" rgba="0.45 0.45 0.45 1"/>
        </default>
      </default>
    </default>
  </default>

  <asset>
[... 249 lines cut ...]
    <jointvel name="ur5e/elbow_joint_vel" joint="elbow_joint"/>
    <jointvel name="ur5e/wrist_1_joint_vel" joint="wrist_1_joint"/>
    <jointvel name="ur5e/wrist_2_joint_vel" joint="wrist_2_joint"/>
    <jointvel name="ur5e/wrist_3_joint_vel" joint="wrist_3_joint"/>

    <torque name="ur5e/shoulder_pan_joint_torque" site="shoulder_pan_joint"/>
    <torque name="ur5e/shoulder_lift_joint_torque" site="shoulder_lift_joint"/>
    <torque name="ur5e/elbow_joint_torque" site="elbow_joint"/>
    <torque name="ur5e/wrist_1_joint_torque" site="wrist_1_joint"/>
    <torque name="ur5e/wrist_2_joint_torque" site="wrist_2_joint"/>
    <torque name="ur5e/wrist_3_joint_torque" site="wrist_3_joint"/>

    <force name="ur5e/wrist_force" site="attachment_site"/>

    <framepos name="gripper/pinch_pos" objtype="site" objname="pinch"/>
    <framequat name="gripper/pinch_quat" objtype="site" objname="pinch"/>
    <framelinvel name="gripper/pinch_vel" objtype="site" objname="pinch"/>
    <jointpos name="gripper/finger_pos" joint="right_driver_joint"/>
  </sensor>
</mujoco>

```
</details>

It simulates perfectly fine for me. Can you check that your merged XML is similar and has all the sections as above?

ChristosPeridis, 2023-12-20:
Hello dear Kevin,

Thank you very much for your immediate response and the  supplied XML file. I will study it thoroughly and compere it with the one I have created for the UR10e and 2f85 gripper integration. I will be keeping you updated regarding my progression on these works.

Thank you very much for the valuable help and support!!!

Kind regards,

Christos Peridis

kevinzakka, 2023-12-25:
Hi @ChristosPeridis, did you figure out your issue?

kevinzakka, 2024-01-03:
Going to close this, feel free to re-open if you fail to find the issue with your model.

closed 2024-01-03 by kevinzakka

vnazary, 2025-11-11:
Hi there, 
It sounds cool if you manually copy and paste the contents of one xml file to another one and make the robot and the gripper merged. However, it is a time consuming process when you want to change the gripper or the robot. What is the best practice to attach or detach a gripper from a robot without touching the xmls?