google-deepmind/mujoco_menagerie #40: Attaching the RobotIQ 2f85 gripper to the UR10e robotic arm (Reopen of issue #37)

ChristosPeridis, 2024-01-10:
With 2f85 gripper

When combing the tow xml files for the UR10e and the 2f85 gripper, the 2f85 gripper appears to be glitching in the simulation.

Hello dear @kevinzakka,

I hope you and everyone in your family are all well and healthy !!! I am also wishing a happy New Year !!! All the best for you and your family, but most importantly Good Health, Love and Happiness!!! I am creating this new issue as a continuation of the original #37 that I had opened. I do apologise for my late response, however my works on the project had been postponed due to the Christmas Break. I have tested the .xml file you supplied me with and describes the integration model of the UR5e robotic arm and the 2f85 gripper, with the script I have developed. Initially I received an error regarding the setting of _**'kv'**_ parameter under the _'default class position'_ inside the tag '_position'_. The system was not able to recognise the parameter kv, so I deleted it. I am using _**MuJoCo version 2.3.3**_. After applying the abovementioned change in the supplied xml file, I was able to run it with my Python script. However, I did not get the behaviour I was expecting from the script. Initially it start working as expected, slowly closing and opening the gripper fingers. After a while though, it starts making very abrupt movements, and the fingers remain mostly in a semi-open-closed position, while it looks like it rotates the last actuator of the robotic arm, despite me specifically controlling the actuator that is responsible for the gripper fingers. I achieve this functionality by selecting the actuator  with the id that matches the desired name given to the actuator, in this case the name is _'finger_actuator_. I am providing videos of the runs for the tow xml files run by my manipulation script. One video for the  xml you have provided me with, and implements the integration of the UR5e with the 2f85 gripper, and one vide with the xml I have developed and implements the UR10e integration with the 2f85 gripper.

https://github.com/google-deepmind/mujoco_menagerie/assets/34765945/d8939df5-76d3-409e-ba44-85dec85dd44d

https://github.com/google-deepmind/mujoco_menagerie/assets/34765945/5ca22e17-271a-431b-bb03-5a382328e1f4

 
I am also uploading the gripper manipulation script that I have developed.

[ur5e_2f85_gripper.zip](https://github.com/google-deepmind/mujoco_menagerie/files/13894189/ur5e_2f85_gripper.zip)

I have also further compered the two xml files, especially in the part of the code where the attachment of the two bodies happen through the attachment site.

[image]

The main differences that I can spot is the existence on extra _site_ in the ur5e _2f85 combo which is called _'wrist_3_joint_, and the creation of a body with _name_ '_attachment'_, where in my implementation the extra site regarding the wrist 3 does not exist, and I directly attach the base mount of the gripper on the site _'attachment_site'_ itself, without making an attachment body.  Further differences regarding the structure of the two xml files are that, where possible, I created separate tags for the UR10e and the 2f85. For example I have created two different _'asset'_ tags, one for the UR10e and one for the 2f85. Furthermore, in the default section I have created a _"mother_class"_ called  _'ur10e_2f85'_ which includes the two separate subclasses called _'ur10e'_ and _'2f85'_ respectively.

I thank you very much in advance dear @kevinzakka  for your valuable help and support!!!

Kind regards,

Christos Peridis

ChristosPeridis, 2024-01-18:
Hello dear @kevinzakka ,

How are you doing ? I hope you and everyone in your family are doing well! I am sending this message as a kind reminder regarding the new issue that I have made in order to re-open the older issue #37 . Furthermore, I would like to report an other difference that I have spotted between the two integration implementations. In the UR5e integration with the 2f85 XML file that you supplied me with the actuators  for the UR5e are of _type **<position>**_ , whilst in the original UR5e XML file in the MuJoCo menagerie repository are defined as **_<general>_**. In my implementation of the integration of the UR10e with the 2f85 I have used the **_<general>_** type of actuators, as they have been defined in the original XML file for the UR10e in the MuJoCo menagerie repository.

Thank you very much for the valuable help and support!

Kind regards,

Christos Peridis

kevinzakka, 2024-02-09:
Hi @ChristosPeridis, a few things:

- Please refrain from opening issues to point to an existing issue.

- > Initially I received an error regarding the setting of 'kv' parameter

kevinzakka, 2024-02-09:
Hi @ChristosPeridis, please refrain from opening issues to point to an existing issue.

> Initially I received an error regarding the setting of 'kv' parameter...

This is a feature that was made available in version 3.1.0. Please upgrade your MuJoCo setup if you can.

Regarding the main issue, I would recommend you attach a zip file with a fully loadable XML file so that I can debug your issue further.

Lastly, can you let me know how exactly you're running your simulation? Are you using the simulate binary? `dm_control`? Or is this the python bindings + `mujoco-python-viewer`?

ChristosPeridis, 2024-03-24:
Hello dear @kevinzakka,

How are you doing ? I hope that you and everyone in your family are all well and healthy! I do apologize for my  my late reply. My works on this project have been postponed at the moment, due to other higher priorities in my lab. 

Regarding the 'kv' value issue, I am developing Reinforcement Learning environments for these robots, following the principals of the Farama's-Foundation Gymnasium-Robotics API (version 1.2.2) . In this version of the Gymnasium-Robotics API they had been using MuJoCo version 2.3.3, and this is the version that I have been using through the development of this project. Because the environments, have not yet been finished in design, I would avoid make any changes to the versions of the different Python libraries that have been used. Once, their initial design has been finished and validated that they are working correctly, its is of my top priorities, to update the code to the latest versions of the different Python libraries. 

Regarding the main issue, I have manged to fix it by changing the integrator to "RK4" from "implicitfast" .

`<option impratio="10" integrator="RK4"/>`

https://github.com/google-deepmind/mujoco_menagerie/assets/34765945/204cdeb8-44cc-4555-ba78-0a8e9aeb5ffe

Here, I do supply you with the raw XML file that I have develop for the integration of the UR10e robot with the RobotIQ_2f85 gripper. Please fill free to investigate it further and give suggestions on how I could potentially improve it further.

[ur10e_2f85_pose_scene.zip](https://github.com/google-deepmind/mujoco_menagerie/files/14736623/ur10e_2f85_pose_scene.zip)

Thank you very much for your valuable help and support!!!

Kind regards,

Christos Peridis

ChristosPeridis, 2024-04-03:
Hello dear @kevinzakka,

I hope you and everyone in your family are well and healthy!!! My apologies on not clearly specifying how I run the simulation. I have installed MuJoCo 2.3.3 with the corresponding version of the Python Bindings. I am visualizing the environment using the MuJoCo Renderer provided by the Farama-Foundation Gymnasium API (version 0.29.0), under `Gymnasium /gymnasium/envs/mujoco/mujoco_rendering.py. 

Thank you very much for your valuable help and support!!!

Kind regards,

Christos Peridis

pnaeemi, 2024-04-23:
where can i download this robot with gripper xml files and run into MUJOCO simulator?

kevinzakka, 2024-06-04:
Have been unable to reproduce this issue and I believe the UR5e and Robotiq models attached behave as expected.

closed 2024-06-04 by kevinzakka