google-deepmind/mujoco_menagerie #45: Barkour contact parameters

Nate711, 2024-03-02:
This model might be missing the `impratio` setting which is important to allow soft foot contact but limit slipping. The other quadrupeds set it to 100. I also noticed the `solimp="0.015 1 0.031"` line is setting the d_width (compression distance) to 31mm which I think is bigger than Barkour's foot.  

https://github.com/google-deepmind/mujoco_menagerie/blob/66080c4bb74969c4508dc6663e0f9f5242ee9e13/google_barkour_vb/barkour_vb_mjx.xml#L14

Reference: https://github.com/google-deepmind/mujoco_menagerie/blob/66080c4bb74969c4508dc6663e0f9f5242ee9e13/unitree_a1/a1.xml#L4

kevinzakka, 2024-03-03:
@Nate711 `impratio` is not currently available in MJX (will be in the upcoming release).

Adding @btaba and @yuvaltassa to chime in about the solimp parameter.

btaba, 2024-03-04:
Hi @Nate711 , indeed `impratio` helps limit slipping and will be available in the next MuJoCo release, hence why it is omitted from the vB XML.  The foot diameter for the vB is 35mm, similar to the A1 at 40mm, which uses the same `solimp` param.

Nate711, 2024-03-07:
The regular barkour also doesn't have impratio FYI. https://github.com/google-deepmind/mujoco_menagerie/blob/66080c4bb74969c4508dc6663e0f9f5242ee9e13/google_barkour_vb/barkour_vb.xml

Also I think the 31mm d_width for the A1 was copied blindly from anymal https://github.com/google-deepmind/mujoco_menagerie/blob/66080c4bb74969c4508dc6663e0f9f5242ee9e13/anybotics_anymal_c/anymal_c.xml#L20

closed 2024-06-01 by copybara-service[bot]