google-deepmind/mujoco_menagerie #47: New Model Spot

louislelay, 2024-03-09:
If you would like to see a new model added to the Menagerie collection, please provide us with the following information _before_ sending a pull request. (Ideally, before you start working on converting or tuning the model for MuJoCo.)

**URL for the existing model source (e.g. URDF)** [RS_Description](https://github.com/SoftServeSAG/spot_simulation/tree/spot_robomaker/simulation_ws/src/rs_description)

**License terms of the upstream model** BSD 3-Clause License

**What do you think is required to convert and tune this model for MuJoCo?** 

**Are you proposing to help develop and maintain the new model?** I wanted to learn how to use the MJCF format, so I created a [first version](https://github.com/louislelay/Spot-Mujoco) based on the urdf and inspired by the structure of the Anymal C xml file. I'm having a problem with the texture, Spot appears yellow where it shouldn't. If you know how to solve this problem, I'd really appreciate hearing your suggestions. And if there are any other problems, I look forward to hearing about them too!

kevinzakka, 2024-05-31:
Hi @louislelay, spot model will be released hopefully tomorrow!

kevinzakka, 2024-05-31:
Spot (and arm) are in Menagerie as of 65f5e9ae03108ed1655c49faef39054c273d75a4!

louislelay, 2024-05-31:
Very cool, thank you!