google-deepmind/mujoco_menagerie #48: about mujoco model

ccchen091, 2024-03-15:
Hello, your shadow_hand model needs mujoco 2.2.2 or later, but mujoco-py is only updated to version 2.1, so I would like to ask how to import your model into mujoco and drive this model

yuvaltassa, 2024-03-15:
We don't support `mujoco-py`, exactly for these reasons.
https://mujoco.readthedocs.io/en/latest/python.html#migration-from-mujoco-py

closed 2024-03-15 by yuvaltassa

ccchen091, 2024-03-16:
> We don't support `mujoco-py`, exactly for these reasons. https://mujoco.readthedocs.io/en/latest/python.html#migration-from-mujoco-py

So if I need to implement control over the Mujoco model, the first thing I need to do is download the latest version of Mujoco and then install dm_control, right? Looking forward to hearing from you.

yuvaltassa, 2024-03-16:
TBH dm_control is also on its way out. I would recommend working directly with the Python bindings. Unless you need very specific features of dm_control, it's probably better to just roll your own