google-deepmind/mujoco_menagerie #49: Loading Cassie model into Brax

Icepomel0, 2024-03-23:
**Which model is the issue affecting?**
agility_cassie

**What is the issue?**
When I use the code:
`from brax.io import mjcf
path = '~/agility_cassie/scene.xml'
sys = mjcf.load(path)`
it returns such error message:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/icepomelo/.local/lib/python3.10/site-packages/brax/io/mjcf.py", line 512, in load
    mj = load_mjmodel(path)
  File "/home/icepomelo/.local/lib/python3.10/site-packages/brax/io/mjcf.py", line 506, in load_mjmodel
    mj = mujoco.MjModel.from_xml_string(xml, assets=assets)
ValueError: XML Error: Schema violation: unique element 'inertial' found 2 times

Element 'body', line 0`
**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**
I looked up the model to find the duplicate 'inertial' define for the model but I did not find anything. How to fix such issue? Thanks in advance!

kevinzakka, 2024-03-26:
Hi @Icepomel0, can you post this on the brax github?

Icepomel0, 2024-03-26:
Hi Kevin,

Thanks for the reply. I think the issue is not from Brax but it is from MJX, where it does support some of the existing function of origin MuJoCo. I tried to load the model google_barkour by loading a specific defined in MJX one and it does cost any issue. Therefore I think the issue was raised from the model compatibility itself.

btaba, 2024-03-27:
Hi @Icepomel0 

The model is not compatible with MJX, feel free to make a PR for a compatible version. Here is what I tried:

```python
import mujoco
from mujoco import mjx
m = mujoco.MjModel.from_xml_path('agility_cassie/scene.xml')
mx = mjx.put_model(m)

---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
Cell In[5], line 1
----> 1 mx = mjx.put_model(m)

File ~/mujoco/mjx/mujoco/mjx/_src/io.py:91, in put_model(m, device)
     88   raise NotImplementedError('tendons are not supported')
     90 if (m.geom_condim != 3).any() or (m.pair_dim != 3).any():
---> 91   raise NotImplementedError('only condim=3 is supported')
     93 # check collision geom types
     94 for (g1, g2, *_), c in collision_driver.collision_candidates(m).items():

NotImplementedError: only condim=3 is supported
```

Now regarding the brax issue, `brax.io.mjcf.loads` does a "fuse bodies" step before loading, which is annoying. @erikfrey

You can do this in brax instead:

```python
import mujoco
from brax.io import mjcf
m = mujoco.MjModel.from_xml_path('agility_cassie/scene.xml')
mjcf.load_model(m)
```

Icepomel0, 2024-03-27:
Hi @btaba 

Thanks for the help and declearation. Is there any pointer for how to edit the model to make it compatIble to MJX?

kevinzakka, 2024-03-27:
Hi @Icepomel0, you can take a look at the [MJX page](https://mujoco.readthedocs.io/en/latest/mjx.html) in the docs and an example of [a MuJoCo model](https://github.com/google-deepmind/mujoco_menagerie/blob/main/google_barkour_vb/barkour_vb.xml) vs [its MJX counterpart](https://github.com/google-deepmind/mujoco_menagerie/blob/main/google_barkour_vb/barkour_vb_mjx.xml) in the [Google Barkour vB](https://github.com/google-deepmind/mujoco_menagerie/tree/main/google_barkour_vb) description of this repo.

Icepomel0, 2024-03-27:
Hi @kevinzakka 

Thanks for your help. Will see and try to make it compatible and contribute it back.

closed 2024-04-09 by kevinzakka