google-deepmind/mujoco_menagerie #54: Minimal pick up task using the Franka Panda arm not working as expected

fahimfss, 2024-05-03:
Hi,

I am trying to create a minimal pick-up a box object simulation using the Franka arm in Python. I have managed to do it using the gymnasium Franka arm [(kitchen_franka)](https://github.com/Farama-Foundation/Gymnasium-Robotics/tree/main/gymnasium_robotics/envs/assets/kitchen_franka). The full code can be found [here](https://github.com/fahimfss/mujoco_tests/tree/main/kitchen_panda). Here's the output video: 

https://github.com/google-deepmind/mujoco_menagerie/assets/8725869/b4e59e7b-2abb-45e9-ba39-a9068ecb6f06

However, when I tried to do the same thing with the menagerie Franka arm, it could not lift up the box object. The code can be found [here](https://github.com/fahimfss/mujoco_tests/tree/main/menagerie_panda). Here's the output:

https://github.com/google-deepmind/mujoco_menagerie/assets/8725869/8e9eaabe-6eca-467c-be5e-3578b28ee5d3

I saw a few posts with a similar issue ([this](https://github.com/google-deepmind/mujoco/issues/1592), [this](https://github.com/google-deepmind/mujoco/issues/239)). However, I was not able to solve it in my implementation. 

Any help is appreciated. Thank you.

yuvaltassa, 2024-05-05:
See new section, hope this helps.
https://mujoco.readthedocs.io/en/latest/modeling.html#preventing-slip

closed 2024-05-05 by yuvaltassa

fahimfss, 2024-05-09:
Thank you @yuvaltassa. It turned out that the fingers of Franka's arm were not applying enough force. Updating this line: 

```
<general class="panda" name="actuator8" tendon="split" forcerange="-100 100" ctrlrange="0 255"
      gainprm="0.01568627451 0 0" biasprm="0 -100 -10"/>
```

to this: 

```
<general class="panda" name="actuator8" tendon="split" forcerange="-100 100" ctrlrange="0 255" gear="20 0 0 0 0 0"
      gainprm="0.3137254902 0 0" biasprm="0 -100 -10"/>
```

in the [panda.xml](https://github.com/google-deepmind/mujoco_menagerie/blob/main/franka_emika_panda/panda.xml) file mostly solved the issue for me. Added `<option  impratio="10" noslip_iterations="3"/>` to slow down slipping.