google-deepmind/mujoco_menagerie #55: Robot Arms + Hands Coming Soon?

jonzamora, 2024-05-07:
Hi folks,

I'd like to know if there are plans to add a `Robot Arms + Hands` section to the repository?

(e.g. `xArm7 + Shadow E3M5`, `Panda + Allegro V3`, etc.)

I ask because i'd like to contribute to this effort, and am wondering if any authors of the library have this as part of the Menagerie's development roadmap?

CC: @kevinzakka , in case you might know : )

kevinzakka, 2024-05-07:
Hi Jon!

Our goal is to provide canonical descriptions for each robot *separately*, so we don't have plans to create XMLs for robot combinations.

* Use [PyMJCF](https://github.com/google-deepmind/dm_control/tree/main/dm_control/mjcf#pymjcf) to combine robot models (note that this requires `dm_control` install) and I've put an example in the [FAQ](https://github.com/google-deepmind/mujoco_menagerie/blob/main/FAQ.md) for how to do this.
* Manually combine XMLs (painful so not recommended)

jonzamora, 2024-05-07:
Thanks for the timely response, Kevin! This looks great, I will check it out. 

1 more question: If I plan to open-source these robot combinations for the community, is there a possibility it could be added in the `README.md` of this repository under a "Third-Party" heading, similar to https://github.com/google-deepmind/mujoco?tab=readme-ov-file#third-party-bindings?

kevinzakka, 2024-05-13:
@jonzamora Sure :)

closed 2024-05-13 by kevinzakka