google-deepmind/mujoco_menagerie #58: MJX model versions

peterdavidfagan, 2024-05-17:
**Which model is the issue affecting?**

All models.

**What is the issue?**

It would be nice if model releases distinguish include mjx compatible versions. Models that contain tendons for instance are not supported in mjx, right now users have to alter these published models to use them with mjx, it would be nice if there was official support and/or versioning for mjx compatible models. 

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

I have been trying to use the robotiq_2f_85 model with mjx but I have encountered the following error:

```
 File "/home/peter/.cache/pypoetry/virtualenvs/mujoco-robot-environments-hs2qeFWj-py3.10/lib/python3.10/site-packages/mujoco/mjx/_src/io.py", line 69, in put_model
    raise NotImplementedError('tendons are not supported')
NotImplementedError: tendons are not supported
```

kevinzakka, 2024-05-17:
Hi @peterdavidfagan, we do have MJX-compatible models for certain robots, e.g. see anymal C, Unitree Go2 and Google Barkour robots. Not all models are simulate-able in MJX due to lack of feature parity, but having an MJX-version for all robots that support it is on our roadmap and we are actively accepting community contributions! I'll add an MJX column to the README table to make this more visible.

peterdavidfagan, 2024-05-19:
Ah ok understood, thanks @kevinzakka for clarifying this.

closed 2024-05-19 by peterdavidfagan

kevinzakka, 2024-05-21:
@peterdavidfagan I just pushed a commit that improves the README. There should now be an MJX column in the main table telling you if an MJX-compatible model is available. Cheers!

peterdavidfagan, 2024-05-21:
Thanks @kevinzakka.