google-deepmind/mujoco_menagerie #66: General question about creating an MJCF file

emrecemal, 2024-06-22:
I want to create an MJCF file for my bipedal robot that I designed in Onshape. I used the [onshape-to-robot](https://github.com/Rhoban/onshape-to-robot) tool to obtain the URDF file for the robot. From what I understand, I need to manually create the MJCF file based on the URDF. However, I find the process quite complex as the [XML Reference](https://mujoco.readthedocs.io/en/stable/XMLreference.html#) is very detailed, and I'm hesitant to delve deeper into it. Can you provide me with some guidelines or suggestions on how to convert my robot more conveniently?

yuvaltassa, 2024-06-24:
Did you see [this section](https://mujoco.readthedocs.io/en/latest/modeling.html#urdf-extensions)?

emrecemal, 2024-06-24:
I saw the section and tried that part. I added the following line to the URDF file: 

```
<mujoco>
    <compiler meshdir="./meshes/" discardvisual="false"/>
</mujoco>
```

Then, I dragged the URDF file into the Mujoco App and saw my robot there. However, when I save the MJCF file and open it, I can't see the inertia values. Do I need to modify it manually?

yuvaltassa, 2024-06-24:
That is surprising. URDFs always have explicit inertial clauses, so they should always get saved...

Would you mind making a minimal URDF (no meshes or other external assets) that shows this issue?

yuvaltassa, 2024-06-24:
Hold on, can you set [fusestatic](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-fusestatic) to false? It is on by default for URDF, so these bodies get fused with the world body, which has no inertia...

emrecemal, 2024-06-24:
Here is the revised text:

I have achieved the desired result! Thank you very much for your assistance. Regarding the model transfer, here are the following steps:

1. I obtained the `scene.xml` file from one of the models in the repository and made the necessary import for my robot. However, I noticed that the model has collisions with the ground despite the absence of collision geometry definitions in the new XML file. Is this the expected behavior? Do I need to define the collision geometries?
2. I inserted the `<freejoint/>` line just below my robot's trunk to ensure proper floating base dynamics.
3. I added lighting to track my robot.
4. I manually defined the sensors and actuators.

Do I need extra steps to get a complete model?

kevinzakka, 2024-08-07:
@emrecemal Looks reasonable to me.

closed 2024-08-07 by kevinzakka