google-deepmind/mujoco_menagerie #69: Boston Dynamics Spot - unexpected behaviour in MJX

goncalog, 2024-07-04:
**Which model is the issue affecting?**
Boston Dynamics Spot

**What is the issue?**
Adding the Spot's model to a MJX notebook and rendering a short rollout (e.g. with `state = jit_reset(jax.random.PRNGKey(0))` and `state = jit_step(state, ctrl)`) results in a very unexpected behaviour better described by the first 3 frames (the video remains black for the remainder of the rollout):
_Frame 1 -_
[image]

_Frame 2 -_
[image]

_Frame 3 -_
[image]

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**
It's likely this is me doing something wrong but I had used the exact same setup with the humanoid, robotis op3, and unitree h1 models, and this had never happened.

Any ideas of what may be causing it or what i may try to solve it? Thank you.

scott-yj-yang, 2024-07-15:
Hi goncalog,

I have this similar issue when using the CPU version of the mujoco xml file in MJX, and the rendering seems like the agent is falling through the ground. A temporary fix we found is add `solimplimit="0 .99 .01"` to each of the joint attribute, and it fixed the contact issue in MJX. However, I am not completely sure whether there is any unintended side effect of this change. It will be really helpful if there is a guide of how to convert CPU MuJoCo XML file into a MJX compatible formats.

Scott

kevinzakka, 2024-08-07:
Out of curiosity, did you remove the freejoint in the base?

kevinzakka, 2025-01-22:
Going to close due to inactivity. I haven't been able to reproduce this issue.

closed 2025-01-22 by kevinzakka