google-deepmind/mujoco_menagerie #73: Collision Issue in `spot_arm.xml` Model's `arm_link_sh0` Link

jc-bao, 2024-07-16:
**Which model is the issue affecting?**

`spot_arm.xml`

**What is the issue?**

The model's link `arm_link_sh0` has a collision with other links. 

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

Here is the contact force: 

[image]

As you can see, the arm base has a large contact force, making the arm immovable. By removing the following collision geometries:

```xml
<geom class="collision" mesh="arm_link_sh0_base_coll" />
<geom class="collision" mesh="arm_link_sh0_left_motor_coll" />
<geom class="collision" mesh="arm_link_sh0_right_motor_coll" />
```

The arm can move freely.

kevinzakka, 2024-07-16:
@jc-bao I cannot reproduce this locally. We [carefully checked and excluded](https://github.com/google-deepmind/mujoco_menagerie/blob/main/boston_dynamics_spot/spot_arm.xml#L253-L258) spurious collisions.

What MuJoCo version are you using?

jc-bao, 2024-07-16:
Hi @kevinzakka, I am using `3.1.6` under Ubuntu 22.04. Let me double check the original model and see what's wrong. My modification is to fix the base and only test the `diffik` in your repo. Thanks for the prompt response!

kevinzakka, 2024-07-16:
@jc-bao By "fix base", do you mean removing the `freejoint`? If the answer is yes, then that explains the behavior you are seeing. The reason is that MuJoCo excludes collisions between parent and child bodies by default, but this exclusion is not applied if the parent is a static body (aka world or dof-less descendant of world).

kevinzakka, 2024-07-16:
To fix your issue, you can add `<exclude body1="body" body2="arm_link_sh0"/>` to the exclude clause of the XML.

jc-bao, 2024-07-16:
Oh that make sense. I can confirm that the original model works fine. After exclude the contact, it works perfectly. Thanks!

closed 2024-07-16 by jc-bao

yuvaltassa, 2024-07-17:
For future reference, this is documented here

https://mujoco.readthedocs.io/en/3.2.0/overview.html#surprising-collisions