google-deepmind/mujoco_menagerie #75: Forward Kinematics of UR5e

karthyyy, 2024-07-17:
Hi,

  I am simulating UR5e model from the repo. I could implement the Forward and Inverse Kinematics of the manipulator correctly as per the MuJoCo model and its reference frames. However, when trying to correlate the transformation matrix from the MuJoCo (formulated from the xmat and xpos values), with that generated from the D-H parameters as given in the following link, I am not getting identical results. The translation values are same but with the axis changed, whereas the rotation matrix cannot be correlated.

https://www.universal-robots.com/articles/ur/application-installation/dh-parameters-for-calculations-of-kinematics-and-dynamics/

  It was found that the world frame and base frame are having a rotation. So, I included correction for the same but still I couldn't get identical transformation matrix from MuJoCo and that from the excel sheet in the UR site.  (https://s3-eu-west-1.amazonaws.com/ur-support-site/45257/DH-Transformation.xlsx).

  Can someone help me with the same ?

Regards
  Karthik

kevinzakka, 2024-07-17:
Hi @karthyyy, can you try removing the quaternion at the base [here](https://github.com/google-deepmind/mujoco_menagerie/blob/main/universal_robots_ur5e/ur5e.xml#L64) and let me know if that solves your problem?

karthyyy, 2024-07-18:
Hi @kevinzakka,

  Thanks for the response. I have removed the same but the issue persists. I will explain in detail. Following are the results with "quat" removed from the base.

My joint angles are [3*pi/2, -pi/2, pi/2, 3*pi/2, 3*pi/2, 0] as shown below

[image]

1) The transformation matrix between world frame and base frame in MuJoCo is Identity matrix as expected.

[image]

2) The first transformation matrix (T_01, Joint-1 frame with respect to Base frame/0) from UR excel file is 

[image]

3) In MuJoCo, the Rmatrix/Quaternion etc are with reference to world frame. In this case world frame and base frame are same. So the first transformation matrix (T_W1) from MuJoCo is given as follows. If we see the rotation matrix portion, this is the matrix generated due to a rotation of 3*pi/2 (or -pi/2) as per rotation matrix definition. So its correct.

[image]

The code I used for estimating transformation matrix in MuJoCo is given as
 
```
### World to Joint-1

T_W1_sim_pos   =  data.body('shoulder_link').xpos.copy()
T_W1_sim_quat  =  data.body('shoulder_link').xquat.copy()
T_W1_sim_xmat  =  data.body('shoulder_link').xmat.copy()

r = R.from_quat([T_W1_sim_quat[1], T_W1_sim_quat[2], T_W1_sim_quat[3], T_W1_sim_quat[0]])
print(r.as_euler('xyz', degrees=False))

T_W1_mj = Matrix(np.eye(4))
T_W1_mj[:3,:3] = TR10i(T_W1_sim_xmat.reshape(3,3))
T_W1_mj[:3, 3] = T_W1_sim_pos
print(T_W1_mj)
```

 
   I understand that the transformation matrix is different for Standard D-H and Modified D-H Parameters. U-R excel file follows Standard D-H parameters. There is an online literature analyzing UR5 with modified D-H parameters. If we take modified D-H parameters, still the transformation matrices are different from MuJoCo model.
 
  I believe for D-H parameters, the joint rotation axis is always 'Z' but in the xml file the default joint axis is 'Y'. What may be the reason ?
 
  Since I am relatively new to this, I believe I am wrong somewhere but couldn't figure it out. Any help will be much appreciated.

Regards,
   Karthik

loerting, 2024-08-06:
I do not have a solution, but a similar problem in my issue #85.

kevinzakka, 2024-08-07:
Hi @karthyyy, could you let me know if the [code example](https://github.com/google-deepmind/mujoco_menagerie/issues/85#issuecomment-2272451871) I posted in #85 is helpful?

karthyyy, 2024-08-07:
Hi @kevinzakka,

   I had tried Standard and Modified D-H parameters for UR5e but was not successful. So I manually extracted the transformation matrices of each joints and found the cumulative matrix. The method now holds good for FK, IK and even Jacobian computation and served my purpose.

   I will try the code mentioned in the other thread for UR5e and get back, as it gives a standard method which can be extended to other manipulators.

Regards
   Karthik

kevinzakka, 2024-09-18:
Going to assume this has been resolved. Feel free to re-open if you encounter further issues.

tlpss, 2024-12-18:
@kevinzakka  sorry to reopen this but I was wondering why this rotation is included in the base body?

 The base frame of the mujoco model  is now not the actual base frame as used by the UR controller I think. (Implying that for a joint configuration, the pose obtained from the controlbox differs from the one obtained in mujoco). 

The [original URDF file](https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_e_description/urdf/ur5e.urdf.xacro#L364) states that a rotation around the z axis over -pi was needed to convert from the ROS frame convention (the `base` element in the URDF)  to the UR controlbox (the `base_link` element in the URDF)

I think it would be easier if the base frame of the xml is consistent with the base frame of the robot controlbox?

This would be solved by setting the quat orientation of the `base` body  to `0,0,0,-1`. 

Following code snippet confirms this:

```
   # check robot pose -joint pairs correspond to IK 
    robot = robot_descriptions.ur5e_mj_description.MJCF_PATH
    robot = mjcf.from_path(robot)

    robot.find("body","base").quat = [0,0,0,-1]
    physics = mjcf.Physics.from_mjcf_model(robot)

    joint_config = np.zeros(6)

    physics.bind(robot.find_all("joint")).qpos = joint_config
    
    # get attachment site pose
    attachment_site = robot.find("site", "attachment_site")
    attachment_site_pose = physics.bind(attachment_site).xpos
    attachment_site_orientation = physics.bind(attachment_site).xmat.reshape(3,3)
    pose = np.eye(4)
    pose[:3, :3] = attachment_site_orientation
    pose[:3, 3] = attachment_site_pose
    print(f"attachment site pose: \r\n {pose}")

    import ur_analytic_ik
    FK_pose = ur_analytic_ik.ur5e.forward_kinematics(*joint_config)
    print(f"forward kinematics pose: \r\n {FK_pose}")

```

[ur_analytic_ik]() has been tested to ensure it's base frame matches the controlbox.

kevinzakka, 2024-12-18:
Hey @tlpss, agreed we should change it just to consistent.

tlpss, 2024-12-19:
@kevinzakka, I can make a PR if you want

kevinzakka, 2024-12-19:
Go for it!

tlpss, 2024-12-19:
I changed the orientations [here](https://github.com/tlpss/mujoco_menagerie/tree/main).

Do you want to add a site/body that corresponds to the frame of the controlbox as well? because now the `base body` 's frame does not correspond to the 'base frame' of the controlbox. (hence the additional 'base_link' in the URDF)

PR #133 (2025-01-10): Match frame of UR robots with base frame used by real robot controlbox
This PR changes the orientation of the `base body` of the UR robots to match the frame of the robot with the frame used by a real robot controlbox.  See [the original URDF]() file for reference, where they also added a separate link that corresponds to the real robot base frame.

This ensures that a given joint configuration corresponds to the same pose for a real robot (expressed in the base frame) and a simulated robot (expressed in the 'attachment site' / world frame). This avoids confusion and allows to use real world t [... cut]
closed 2025-01-10 by copybara-service[bot]