google-deepmind/mujoco_menagerie #81: Add `CHANGELOG.md` and contributors section.

kevinzakka, 2024-07-20:
As title suggests, we should have a changelog that properly documents model changes and community contributions.

louislelay, 2025-02-12:
Hi @kevinzakka,

I'd love to take this on if it's still up for grabs! Here’s what I had in mind:

- Create a CHANGELOG.md file following [Semantic Versioning](https://semver.org/) to properly track changes.
- Add a CONTRIBUTORS.md file to acknowledge all contributors.

That said, I’d be happy to hear your thoughts on this! Let me know if you had something different in mind or any preferences regarding the format or structure.

Thanks!

kevinzakka, 2025-04-20:
Hi @sreshu and @louislelay, sorry for the delayed response.

Yes, this issue is very much still open and would love some help :)

louislelay, 2025-04-20:
Hi @kevinzakka, still very interested! I can start implementing what I suggested earlier if that works for you.

kevinzakka, 2025-04-20:
@louislelay Sounds good to me!

I forgot to mention that I have already started putting `CHANGELOG.md`s in some directories already (e.g. see Unitree G1). How do you want to deal with that? Do we still want a changelog per model and then a global changelog?

louislelay, 2025-04-22:
Sounds good, thanks for the update @kevinzakka!

From what I saw, most models just have a changelog section in their README for now. I was thinking we could:

1. Move those into dedicated `CHANGELOG.md` files inside each model folder.
2. Link to the new changelogs from the READMEs to keep things easy to find.
3. Add a global `CHANGELOG.md` at the root of the repo to track broader updates (e.g. repo structure, CI, naming conventions) and maybe reference the model-specific logs.

Let me know if that works for you — happy to start putting it together!

kevinzakka, 2025-04-22:
Yeah that sounds great @louislelay!

closed 2025-04-23 by kevinzakka