google-deepmind/mujoco_menagerie #82: realsense_d435i simulation - Does the simulation mimic the reality when it comes to the output of the intel realsense

gautam1858, 2024-07-25:
Hi,

I want to know whether the output of the intel realsense camera d435i in the mujoco simulation mimics the output of the actual intel realsense camera d435i ?

jonzamora, 2024-07-29:
From looking at the [XML file](https://github.com/google-deepmind/mujoco_menagerie/blob/main/realsense_d435i/d435i.xml), it seems like the camera is just for looks and does not provide `RGBD` sensor readings. I think you will have to get `RGBD` readings from MuJoCo itself, like they cover in their introductory tutorial found here -> https://github.com/google-deepmind/mujoco

It would be cool though if this RealSense camera model **did** provide `RGBD` observations.

Maybe someone could take that task upon themselves 👀

jonzamora, 2024-07-29:
Something which could be useful: 

If you look at the [ALOHA](https://github.com/google-deepmind/mujoco_menagerie/tree/main/aloha) model, they state "The intrinsic parameters of the simulated cameras are matched with an [Intel RealSense D405](https://www.intelrealsense.com/depth-camera-d405/)."

The lines of code you can reference are here: 

- [Left Wrist Camera](https://github.com/google-deepmind/mujoco_menagerie/blob/441860e128653eead795a41b10d8d189e945a99e/aloha/aloha.xml#L152)
- [Right Wrist Camera](https://github.com/google-deepmind/mujoco_menagerie/blob/441860e128653eead795a41b10d8d189e945a99e/aloha/aloha.xml#L240)

This could at least give you similar `RGB` readings for a D405 camera.

closed 2024-07-29 by yuvaltassa

gautam1858, 2024-08-02:
@jonzamora @yuvaltassa  How can we quantitatively or draw comparison to say that the Intel realsense simulation exactly or closely matches  Intel realsense hardware ?

maheshr24, 2024-12-03:
How do I calibrate the camera in this camera in simulation?