google-deepmind/mujoco_menagerie #83: xArm7 Gripper can't Grasp Objects

jonzamora, 2024-07-28:
**Which model is the issue affecting?** 

- [xArm7](https://github.com/google-deepmind/mujoco_menagerie/tree/main/ufactory_xarm7)

**What is the issue?**

- I'm working on a task where an `xArm7` can grasp and lift a cube with the help of [`mink`](https://github.com/kevinzakka/mink), however, I ran into an issue with the gripper.

https://github.com/user-attachments/assets/a3200082-b9ba-478d-b977-4c763f023c6f

Could someone help me debug the `xArm7` gripper so it can grasp the cube properly?

I tried all the suggestions mentioned in https://github.com/google-deepmind/mujoco/issues/786 and https://github.com/google-deepmind/mujoco_menagerie/issues/54, though my attempts were unsuccessful so far.

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

- My code can be found here https://github.com/jonzamora/mink.
- To reproduce the video above, simply run `mjpython examples/arm_xarm7.py`

kevinzakka, 2024-07-29:
I don't have a lot of bandwidth to fix this right now, but some things that can be done to improve this gripper:

* Custom geom colliders for the finger pads, like in the [Robotiq](https://github.com/google-deepmind/mujoco_menagerie/blob/main/robotiq_2f85/2f85.xml#L46-L53) model. Splitting into 2 boxes per pad for more contacts is even better.
* Adding `<option cone="elliptic" impratio="10"/>`.
* The `forcerange` on the gripper actuator seems excessively high (100 N?)
* Adding `armature` to the joints as [was done with the Robotiq model](https://github.com/google-deepmind/mujoco_menagerie/commit/a9f3e331238e5ca82923dfbeb93f4589af97b011) for a bit more stability.

s1lent4gnt, 2024-12-14:
Hi,
I tested your robot arm implementation (@jonzamora) for grasping tasks, following @kevinzakka's recommendations. The results showed improved grasping performance, but it's still not as effective as the Robotiq hand. Here are the modifications I made and the results obtained:

- Added the following option:

`<option impratio="30" timestep=".002" cone="elliptic" noslip_iterations="3"/>`

- Increased friction for the object:

`<geom name="red_cube" type="box" mass="0.1" pos=".39 0 0.025" size=".025 .025 .025" rgba="1 0 0 1" friction="2" priority="1"/>`

- Added armature to the joints:

`<joint axis="0 0 1" armature="0.2" range="-6.28319 6.28319" frictionloss="1"/>`

Here a video demonstrating the result:

https://github.com/user-attachments/assets/5ef57d67-690f-481f-a2e9-ac45b01a6b1a

Contact forces are not stable, leading to inconsistent performance in xArm7:

https://github.com/user-attachments/assets/2769fa37-7620-49bc-9376-f2aba2457521

Contact forces are stable and uniform across the surface with Robotiq:

https://github.com/user-attachments/assets/c2b8301e-f590-44cb-9c07-65f3cda19acd

kevinzakka, 2024-12-15:
@s1lent4gnt I believe the Robotiq geoms were custom designed with boxes (primitive geoms are more stable). Can you see if doing the same for the xarm helps?

s1lent4gnt, 2024-12-15:
Hi @kevinzakka, I’ve experimented with adding a box geometry primitive to the surface of both fingers. I tested configurations with one pad per finger as well as two pads, following your suggestion. The results are quite good and the contact forces are more stable and uniformly distributed on the surface of the pads, especially with the two-pad setup, which resembles the Robotiq design. Here are my findings:

One pad :

[image]

https://github.com/user-attachments/assets/2eae0b97-e18f-4874-9093-30518f34caed

Two pads :

[image]

https://github.com/user-attachments/assets/c0b6aedb-8d58-4686-b9e3-543726aafe2f

Is it possible to submit a PR for the changes I did?

Thank you for your help!

kevinzakka, 2024-12-15:
Nice job @s1lent4gnt!! Yes, a PR would be excellent!

closed 2025-01-22 by kevinzakka