google-deepmind/mujoco_menagerie #86: Intel Realsense 435i

gautam1858, 2024-08-06:
How can we quantitatively or draw comparison to say that the Intel realsense simulation exactly or closely matches Intel realsense hardware ?

kevinzakka, 2024-08-13:
What intel realsense simulation are you referring to?

gautam1858, 2024-08-19:
Image rendering that is coming out of Intel Realsense camera 435i in simulation

[Left Wrist Camera](https://github.com/google-deepmind/mujoco_menagerie/blob/441860e128653eead795a41b10d8d189e945a99e/aloha/aloha.xml#L152)

[Right Wrist Camera](https://github.com/google-deepmind/mujoco_menagerie/blob/441860e128653eead795a41b10d8d189e945a99e/aloha/aloha.xml#L240)

kevinzakka, 2025-01-22:
1. The wrist cameras on the ALOHA seem to be D405s, not 435i.
2. Looks like they were updated to better match real in 441860e128653eead795a41b10d8d189e945a99e.

closed 2025-01-22 by kevinzakka