google-deepmind/mujoco_menagerie #88: Inquiry about finger position and inertia in wx250s.xml model file

fanshJ, 2024-08-13:
**Which model is the issue affecting?**
[trossen_wx250s/wx250s.xml](https://github.com/google-deepmind/mujoco_menagerie/tree/main/trossen_wx250s)

**What is the issue?**
I am currently examining the wx250s.xml model file and would like to bring to your attention some aspects that might require clarification or further review:
**Finger Position**: In the initial state, there is partial overlap between the left and right fingers, and their movement cannot be controlled.
[image]
**Inertia Values**: The rendering of inertia seems to be somewhat abnormal.
[image]

kevinzakka, 2024-08-13:
Both of these things are inherited from the URDF file which we took from the official Trossen repo (see the [README](https://github.com/google-deepmind/mujoco_menagerie/blob/main/trossen_wx250s/README.md#urdf--mjcf-derivation-steps) for the exact sequence of steps we took).

That being said, you can fix the finger collision problem by defining a keyframe with the fingers in a slightly open or completely open configuration and always initialize to that using `mj_resetDataKeyframe`. We actually have one [defined for you already](https://github.com/google-deepmind/mujoco_menagerie/blob/main/trossen_wx250s/wx250s.xml#L130). For the inertia, toggle "scale inertia" and see what happens.

fanshJ, 2024-08-14:
Thank you so much for taking the time to help me with this issue.  I have solved the finger collision problem by adjusting the position of the finger in the model file. 

For the inertia, I obtained a relatively normal inertia matrix by toggling "scale inertia". However, I've encountered a new concern:
Although the inertia matrix appears normal after my adjustments, it seems that the inertia distribution does not match the shape of the object. There appears to be some misalignment or rotation. Is this behavior expected? Could this discrepancy potentially affect the accuracy of my simulations moving forward?
[image]

I am a beginner in both robotics and MuJoCo. Any guidance or clarification you can provide on this issue would be greatly appreciated. Thank you in advance for your time and assistance.

kevinzakka, 2024-08-15:
As I said above, the inertias were obtained from the official URDF, and in general, manufacturer inertias are pretty sloppy. The right thing to do is to sysID the inertias, but unless you are doing highly dynamic motions (unlikely with this robot arm?), then it shouldn't matter too much.

fanshJ, 2024-08-16:
Thank you very much for your response. I deleted the clause about inertia in the model file, it looks much more normal now.

closed 2024-08-16 by fanshJ