google-deepmind/mujoco_menagerie #97: Adding UMI-Gripper

omarrayyann, 2024-09-25:
Hello, I wanna add the UMI-Gripper (https://umi-gripper.github.io). I imported the URDF and got the joints working. I think I just need to color it correctly before it's ready.

https://github.com/user-attachments/assets/ec3524b0-be5d-4be2-9f3a-887399a058de

yuvaltassa, 2024-09-26:
1. This is awesome!
2. I don't understand the question about mirrors? You want a reflective surface? The answer is yes, you can have [one such surface](https://mujoco.readthedocs.io/en/latest/XMLreference.html#asset-material-reflectance).

omarrayyann, 2024-09-26:
I came across this: ["Only the first reflective geom in the model is rendered as such"](https://mujoco.readthedocs.io/en/latest/XMLreference.html#asset-material-reflectance). Is there no other way to have 2? 

The [UMI gripper](https://umi-gripper.github.io/) has a mirror on each of the sides (trick to increase what it sees and give a virtual depth) as show here:

[image]

[image]

I got one to work but the other does not reflect (probably has to do with the thing i mentioned up).

[image]

[image]

Also, It might be a minor detail to add the mirrors anyways.

kevinzakka, 2024-09-26:
Nice job @omarrayyann !

omarrayyann, 2024-10-29:
Opened a PR for this here: https://github.com/google-deepmind/mujoco_menagerie/pull/106

closed 2025-01-24 by omarrayyann