google-deepmind/mujoco_menagerie #98: ALOHA Mujoco Version Compatibility Issue

Hongyoungjin, 2024-09-25:
**Which model is the issue affecting?**

ALOHA

**What is the issue?**

MuJoCo versions before 3.1.1 does not have actuatorfrcrange attribute, while the ones after 3.1.1, we are getting 'MjModel' object has no attribute 'bvh_geomid' error.

**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

yuvaltassa, 2024-09-26:
What exactly are you doing to get the error `MjModel' object has no attribute 'bvh_geomid`?

Hongyoungjin, 2024-09-30:
We are trying to import AlOHA 2 xml file to the mujoco simulation based on the scripts in  ALOHA 1, which uses Mujoco 2.3.7. However, ALOHA 2 xml file requires the latest mujoco version does not have mujoco.physics attribute (mainly needed in ALOHA 1 scripts). Can you suggest any alternatives for this deprecated attribute  so that we can use ALOHA2 xml file for simulation?

yuvaltassa, 2024-10-14:
`mujoco` never had a `physics` attribute. Do you mean `dm_control.Physics`?

Anyway there is not enough context here. Closing for now.

closed 2024-10-14 by yuvaltassa