google-deepmind/mujoco_menagerie #99: Model collisions

Narsimha-swamy, 2024-10-10:
**Which model is the issue affecting?**
ALOHA2 
**What is the issue?**
I am using Aloha 2 models for real-to-sim teleoperation, and I am able to send data from real robots to robots in the sim and control them, but collisions between objects seem to be off as robots are passing through objects in the scene. Is there a way to know if collisions or contact physics, in general, were implemented in the ALOHA2 XML files?. I uploaded a video of the simulation, can you let me know what seems to be the problem. 

https://github.com/user-attachments/assets/c31de7ff-7134-42fe-858d-80cd26da2efc

 
**Is there any additional context you can provide (e.g., a spec sheet or a URDF to show a value mismatch)?**

btaba, 2024-10-10:
Hi @Narsimha-swamy , the video is empty. AFAIK the base Aloha XML files have good defaults, we have been using these files for some time. I would expect the objects you are adding need their physics parameters tuned. Remember to turn on [MultiCCD](https://mujoco.readthedocs.io/en/stable/XMLreference.html#option-flag-multiccd).

Narsimha-swamy, 2024-10-10:
https://github.com/user-attachments/assets/dfbefd6a-e7b1-4eae-8866-86ee5c67640c

btaba, 2024-10-10:
Which scene are you loading? How to repro?

Narsimha-swamy, 2024-10-11:
Hi, I made a new scene by including the scene.xml in the new script. 
`<mujoco>
    <include file="scene.xml"/>
    
    <worldbody>
       
        
        <body name="box" pos="0 0 0.02">
            <joint name="box_joint" type="free" />
            <inertial pos="0 0 0" mass="0.05" diaginertia="0.002 0.002 0.002" />
            <geom condim="4" solimp="2 1 0.01" solref="0.01 1" friction="1 0.005 0.0001" pos="0 0 0" size="0.02 0.02 0.02" type="box" name="red_box" rgba="1 0 0 1" />
        </body>
    </worldbody>
<option gravity="0 0 -9.8" timestep="0.02" />

</mujoco>`

Narsimha-swamy, 2024-10-11:
I made a new scene by importing the aloha scene to it.
`
<mujoco>
    <include file="scene.xml"/>   
    <worldbody>
       <body name="box" pos="0 0 0.02">
            <joint name="box_joint" type="free" />
            <inertial pos="0 0 0" mass="0.05" diaginertia="0.002 0.002 0.002" />
            <geom condim="4" solimp="2 1 0.01" solref="0.01 1" friction="1 0.005 0.0001" pos="0 0 0" size="0.02 0.02 0.02" type="box" name="red_box" rgba="1 0 0 1" />
        </body>
    </worldbody>

    <option gravity="0 0 -9.8" timestep="0.02" />
</mujoco>
`

Narsimha-swamy, 2024-10-15:
Hi, I included the scene.xml from aloha into new scene and added a cube to it 
`<mujoco>
    <include file="scene.xml"/>   
    <worldbody>
       <body name="box" pos="0 0 0.02">
            <joint name="box_joint" type="free" />
            <inertial pos="0 0 0" mass="0.05" diaginertia="0.002 0.002 0.002" />
            <geom condim="4" solimp="2 1 0.01" solref="0.01 1" friction="1 0.005 0.0001" pos="0 0 0" size="0.02 0.02 0.02" type="box" name="red_box" rgba="1 0 0 1" />
        </body>
    </worldbody>

    <option gravity="0 0 -9.8" timestep="0.02" />
</mujoco>`

btaba, 2024-10-24:
@Narsimha-swamy please lower your timestep to say 0.002. 0.02 is too high

closed 2024-10-24 by btaba

Narsimha-swamy, 2024-10-28:
Hi, I have already tried this with timestep = 0.002, but it still gives out similar results.

btaba, 2024-10-29:
@Narsimha-swamy I can't reproduce your issue, send over a full XML example if you like

Narsimha-swamy, 2024-10-29:
Hi, I made a git [repo](https://github.com/Narsimha-swamy/mujoco_aloha_test.git) for the same, please let me know if you find anything wrong with the implementation.

btaba, 2024-11-21:
Hi @Narsimha-swamy , I loaded your `sim_cube_transfer.xml` and once again the issue is that the timestep is too high. Lowering the simulation timestep to 0.002 fixes any jitter. Also the inertia for the cube is unrealistic, and it rolls around quite easily. Hope that helps!

Narsimha-swamy, 2024-11-21:
Hi, I have one more question regarding image  from mujoco. I am using 
`with mujoco.Renderer(model,height,width ) as renderer:
    renderer.update_scene(data, cam= cam_name)
    image = renderer.render()       
`
with this it takes a min of 0.003sec  to max of 0.01sec  per camera per time step . I am using 5 cameras in my simulation. So, including all processes in the time step along with rendering these five cameras it takes about 0.3 sec/timestep, making the robot operate at a frequency of 3hz (3 timesteps/ second ). For my task I am looking to have a frequency of 50hz. Is there a way to reduce this rendering time. I am using the headless GPU rendering by setting MUJOCO_GL = egl.