google-deepmind/mujoco_menagerie #109: Joint Velocity Limits in UR10e

adawne, 2024-11-12:
Hi, was there any specific reason why the forcerange for the UR10e and the grouping of the actuators are defined in [this](https://github.com/google-deepmind/mujoco_menagerie/blob/main/universal_robots_ur10e/ur10e.xml) way?

I am asking because I noticed that the forcerange of the actuators does not match the values specified in the technical datasheet, even though I understand that MuJoCo is unitless. However, the qvel values (I believe data.qvel is returned in rad/s based on [this reference](https://mujoco.readthedocs.io/en/latest/overview.html#units-are-unspecified), please correct me if I am mistaken) produced when I attempted to control some joints using data.ctrl sometimes exceed the velocity limit specified in the [technical datasheet](https://www.universal-robots.com/media/1802779/ur10e-32528_ur_technical_details_.pdf).

For example, when I tried the following code:
```
  model = mujoco.MjModel.from_xml_path("mujoco_menagerie/universal_robots_ur10e/scene.xml")
  data = mujoco.MjData(model)
  time_datas = []
  qvel_datas = []

  with mujoco.viewer.launch_passive(model, data) as viewer:
      start_time = time.time()
      num = 0
      while viewer.is_running() and time.time() - start_time < 300:
          # active_motors.print_actuator_parameters(model)
          step_start = time.time()
          current_time = time.time() - start_time
          if data.time > 1:
              data.ctrl[1] = -2.5
          time_datas.append(current_time)
          qvel_datas.append(data.qvel[:6].copy())

          mujoco.mj_step(model, data)
          viewer.sync()

          time_until_next_step = model.opt.timestep - (time.time() - step_start)
          if time_until_next_step > 0:
              time.sleep(time_until_next_step)
```
I observed that the joint velocities exceeded the maximum limit of 2 rad/s specified in the datasheet
[image]. 
Do you have any recommendation on how to maintaining the velocity constraints as per the datasheet?

kevinzakka, 2024-11-13:
I believe this was a community contribution, so did not vet the forcerange values. Would you be willing to submit a PR that makes them match the specsheet?

kevinzakka, 2024-11-13:
Hi @adawne, I compare our forcerange values with the "effort" values in the official URDF [here](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_e_description/urdf/ur10e.urdf.xacro) and they seem to match. Am I missing something?

kevinzakka, 2025-01-22:
@adawne Sorry for the delayed response. You are correct that velocity limits don't exit in MuJoCo.

> Should I add damping or rescale the force range to keep it within spec sheet?

There's many ways you can approach this, but I will start by saying that our PD gains are pretty high right now and probably don't match real hardware. You might want to enable gravity compensation, significantly reduce the PD gains and then look at whether clipping or rescaling the ctrl is needed.

closed 2025-01-22 by kevinzakka