google-deepmind/mujoco_menagerie #111: Skydio Yaw control

cisimon7, 2024-11-18:
I'm trying to control the yaw velocity of the skydio drone using the 4 motor actuators, but to no avail. I wonder if something is missing in the mujoco xml model to make it not to respond. I expect when I pass a simple control signal say for example [10, 1, 10, 1] as the actuator signal for it to yaw as the drone falls or rise above, but it does. 

Can anyone point me in the right direction as to why this is happening please?

thowell, 2024-11-20:
Please try this model with [simulate](https://mujoco.readthedocs.io/en/latest/programming/samples.html#sasimulate) and load the [keyframe](https://github.com/google-deepmind/mujoco_menagerie/blob/cfa1c99ab03c51e23f48409b7bbb360c02ed8ef5/skydio_x2/x2.xml#L69) ('Load key' under the 'Simulation' tab). The system should hover in place.

cisimon7, 2024-11-21:
And it does hover fine, but seems you didn't understand my question. I'm asking if it's possible to get the drone to yaw at a fixed height, i.e. rotate about the z-axis at a constant height.

saran-t, 2024-11-21:
The model doesn't implement any higher-lever controller on top of the actual actuators (rotors in this case) that are on the robot.

You'll need to implement your own controller for this.

yuvaltassa, 2024-11-22:
Saran is not wrong, but the answer to the specific question is **yes**, this is both possible and easy with the Skydio, just read the XML:

```xml
  <actuator>
    <motor class="x2" name="thrust1" site="thrust1" gear="0 0 1 0 0 -.0201"/>
    <motor class="x2" name="thrust2" site="thrust2" gear="0 0 1 0 0  .0201"/>
    <motor class="x2" name="thrust3" site="thrust3" gear="0 0 1 0 0  .0201"/>
    <motor class="x2" name="thrust4" site="thrust4" gear="0 0 1 0 0 -.0201"/>
  </actuator>

  <keyframe>
    <key name="hover" qpos="0 0 .3 1 0 0 0" ctrl="3.2495625 3.2495625 3.2495625 3.2495625"/>
  </keyframe>
```

All rotors generate the same downward thrust but different torque, depending on the spin direction. 1 and 4 spin one way, 2 and 3 the other way, at equal thrust all the torques cancel out (ever ask yourself why drones always have an *even* number of rotors? that's why) so if the ctrl vector in "hover" hovers (let's call it `h`), then the vector `h + c*[1, -1, -1, 1]` will rotate stably around the z-axis with `c` controlling the rotation torque. Not sure why this is not obvious.

FWIW these numbers are actually from the manufacturer. Apologies @thowell, I initially thought this was about your MJPC task. This Skydio I am actually more familiar with.

closed 2024-11-22 by yuvaltassa

yuvaltassa, 2024-11-22:
Side note, we should probably put the Skydio's in MJPC, it will be much more realistic