google-deepmind/mujoco_menagerie #113: Add Torobo and Torobo Hand

mamoruoka, 2024-11-21:
If you would like to see a new model added to the Menagerie collection, please provide us with the following information _before_ sending a pull request. (Ideally, before you start working on converting or tuning the model for MuJoCo.)

**URL for the existing model source (e.g. URDF)**
URDF file is not open-sourced yet, but xml file is open-sourced. To test the model, please refer to the README in [torobo_mujoco](https://github.com/TokyoRobotics/torobo_mujoco).
Torobo: https://github.com/TokyoRobotics/torobo_mujoco/blob/main/robots/torobo2/torobo2_with_hand.xml
Torobo Hand: https://github.com/TokyoRobotics/torobo_mujoco/blob/main/robots/hand_v4/hand_v4.xml

**License terms of the upstream model**
BSD 3-Clause License

**What do you think is required to convert and tune this model for MuJoCo?**
The model is already converted for MuJoCo.

**Are you proposing to help develop and maintain the new model?**
Yes

kevinzakka, 2024-11-21:
Hi @mamoruoka, lovely models. Feel free to open a PR!

mamoruoka, 2024-11-22:
@kevinzakka 
Thank you for your replay.
I will open a PR next week!