google-deepmind/mujoco_menagerie #121: Whats the camera position for eye-in-hand in mujoco xarm7?

maheshr24, 2024-12-02:
Whats the camera position for eye-in-hand in mujoco xarm7?

I was trying from so long

kevinzakka, 2025-01-22:
Your question is unclear.

yurekami, 2026-05-01:
Looking at this 514 days later — the question is asking about an "eye-in-hand" camera for `xarm7`, but the `xarm7` model in this repo (`ufactory_xarm7/xarm7.xml` and `ufactory_xarm7/scene.xml`) does not define such a camera. Adding one is a per-user setup choice for the user's own scene, not a property of the curated Menagerie model.

There's also been no follow-up from the original reporter since @kevinzakka asked for clarification. Suggest closing as "out of scope for Menagerie / needs more info." Happy to be wrong if the intent was to request that an eye-in-hand camera be added to the xarm7 model itself, but that would be a separate `new-feature` issue.