google-deepmind/mujoco_menagerie #135: Difference between Mujoco xml and MXJ based xml

FaizalKarim280280, 2025-01-04:
Hello,

I’m trying to use MJX for running parallel environments. However, standard MuJoCo XML files don’t work, and I need to use MJX-compatible variants instead. Could you explain the main differences between standard MuJoCo XML files and MJX variants? Also, is there a tutorial or guide to help convert standard files into MJX-compatible ones?

Thank you!

btaba, 2025-01-04:
Hi @FaizalKarim280280 

See the docs [here](https://mujoco.readthedocs.io/en/stable/mjx.html#performance-tuning) and a few examples [here](https://github.com/google-deepmind/mujoco_menagerie/tree/main/google_barkour_vb#mjx) and [here](https://github.com/google-deepmind/mujoco_menagerie/tree/main/anybotics_anymal_c#mjx). Let us know if that helps.

FaizalKarim280280, 2025-01-06:
Thanks for the resources. 
I noticed that most contacts are disabled in the models, which I also realized while experimenting with the Franka Panda MJCF file. Enabling all contacts slows the simulation significantly, even with a single environment, making it impractical. On the other hand, enabling only specific contacts, like the gripper fingertips, keeps the simulation fast but has a drawback: collisions involving other parts of the arm won’t be detected.

kevinzakka, 2025-01-22:
@FaizalKarim280280 We're continuously adding new MJX models (and in fact, introduced a bunch in [MuJoCo Playground](https://github.com/google-deepmind/mujoco_playground/) last week). I believe you've answered your question. Please feel free to re-open if something is unclear.

closed 2025-01-22 by kevinzakka