google-deepmind/mujoco_menagerie #156: Gripper of Robotiq 2f85 with UR5e do not work as supposed to

Fima2003, 2025-03-28:
I am using UR5e with robotiq 2f85 gripper to pick up a cube off the ground

It works fine in general, but when the gripper has to grab the cube, one of its hand becomes angled, whilst the other is fine(see img)

https://github.com/user-attachments/assets/3ab0dae4-ff5d-448b-b866-6e41b818296a

This is the minimal code to reproduce:
```xml
<!-- scene.xml -->
<mujoco model="ur5e scene">
  <include file="ur5e.xml" />

  <statistic center="0.3 0 0.3" extent="0.8" meansize="0.08" />
  <default>
    <default class="soft">
      <geom type="mesh" contype="1" conaffinity="1" friction="1 0.1 0.1" />
    </default>
  </default>

  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.1 0.1 0.1" specular="0 0 0" />
    <global azimuth="120" elevation="-20" />
  </visual>

  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="512"
      height="512"
      mark="cross" markrgb=".8 .8 .8" />
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" />
    <material name="soft_material" rgba="0.9 0.3 0.3 1" />
    <mesh name="soft_cube" file="soft_cube.stl" scale="0.2 0.2 0.2" />
  </asset>

  <worldbody>
    <light pos="0 0 1.5" directional="true" />
    <geom name="floor" size="1 1 0.01" type="plane" material="grid" />

    <body name="target" pos="0.5 0.15 .35" quat="1 0 0 0" mocap="true">
      <geom type="box" size=".05 .05 .05" contype="0" conaffinity="0" rgba=".6 .3 .3 .2" />
    </body>

    <body name="soft_cube" pos="0.5 0.5 0.3">
      <freejoint />
      <inertial mass="5" pos="0 0 0" diaginertia="0.09 0.01 0.09" />
      <geom name="soft_cube" class="soft" mesh="soft_cube" material="soft_material" friction="1 0.1 0.1" solimp="0.98 0.99 0.001" solref="0.02 1.0" />
    </body>

    <body name="wall" pos="0.4 0 0.1">
      <geom name="wall" type="box" size=".1 .1 .1" />
    </body>
  </worldbody>

  <sensor>
    <!-- <fromto geom1="wrist_3_link" geom2="wall" cutoff="1" /> -->
    <!-- <fromto geom1="wrist_3_link" geom2="floor" cutoff="1" /> -->
    <fromto geom1="right_driver" geom2="soft_cube" cutoff="1"/>
    <!-- <actuatorfrc actuator="fingers_actuator" /> -->
  </sensor>
</mujoco>
```

```xml
<!-- ur5e.xml -->
<mujoco model="ur5e">
  <compiler angle="radian" meshdir="assets" autolimits="true" />

  <option integrator="implicitfast" />

  <default>
    <default class="ur5e">
[... 493 lines cut ...]
        new_target_pose = mink.SE3.from_rotation_and_translation(
            SO3.from_rpy_radians(np.pi/2, np.pi/2, 0),
            np.array([0.51, 0.5, 0.3])
        )

        # Move the arm to the target pose
        if move_arm_to_position(viewer, model, data, configuration, tasks, limits, solver, rate, end_effector_task, target_pose, pos_threshold, ori_threshold, max_iters):
            # Once in position, squeeze the object using sensor feedback
            # Define desired gripper force threshold and sensor index
            gripper_force_threshold = 2e-4
            sensor_index = 0  # Update based on your model's sensor configuration
            squeeze_object(viewer, model, data, configuration, tasks, rate,
                           end_effector_task, gripper_force_threshold, sensor_index)
            # After squeezing, generate a new random target pose
        configuration.update(data.qpos)  # Synchronize configuration with the latest simulation state
        mujoco.mj_forward(model, data)
        move_arm_to_position(viewer, model, data, configuration, tasks, limits, solver,
                             rate, end_effector_task, new_target_pose, pos_threshold, ori_threshold, max_iters)
        print("Finished moving.")

        # Continue running until viewer is closed
        while viewer.is_running():
            viewer.sync()
            rate.sleep()
```

kevinzakka, 2025-03-30:
Hi @Fima2003, can you start by enabling gravity compensation and significantly lowering the PD gains and tell me what happens?

159chenJie159, 2025-07-03:
> Hi [@Fima2003](https://github.com/Fima2003), can you start by enabling gravity compensation and significantly lowering the PD gains and tell me what happens?您好，您能否先启用重力补偿并显着降低 PD 增益，然后告诉我会发生什么？

hello,i have a question about gravity compensation.
based on my observation, the load compensation effect achieved by setting the payload and zero for UR5e and FT300 seems to be bad— I'm not sure if this is reasonable.

With a 0.8 kg payload, changing the robot's posture results in an error of several Newtons.
So I tried reading the ft_raw_wrench for gravity compensation, but the values I got were very large, over 20,000. I'm not sure if this is correct or what the units are.