google-deepmind/mujoco_menagerie #170: Add ORCA hand

ArturoRoberti, 2025-05-05:
If you would like to see a new model added to the Menagerie collection, please provide us with the following information _before_ sending a pull request. (Ideally, before you start working on converting or tuning the model for MuJoCo.)

**URL for the existing model source (e.g. URDF)**

The current MuJoCo description exists in the [orcahand_description](https://github.com/orcahand/orcahand_description) repo

**License terms of the upstream model**

We are using an MIT license.

**What do you think is required to convert and tune this model for MuJoCo?**

As mentioned above, the current MuJoCo description exists in the [orcahand_description](https://github.com/orcahand/orcahand_description) repo.

In the future, a left ORCA hand will be added too.

**Are you proposing to help develop and maintain the new model?**

Definitely

kevinzakka, 2025-05-05:
Would be great to have the ORCA hand here, feel free to make a PR!

LinFei83, 2025-09-18:
Hello,

I am currently working on a project that aims to perform a block manipulation (rotation) task using the ORCA hand within the mujoco_playground framework.

For this, I am using the MJCF (XML) model file provided in the official orcahand_description GitHub repository. Unfortunately, I've run into a critical issue when loading the model: during MuJoCo's JIT compilation process, the memory consumption steadily increases, eventually exhausting all system resources and causing the system to freeze or crash entirely.

Given this problem, I would like to inquire if there are any plans or progress on integrating the ORCA hand into the official mujoco_menagerie library. I believe an officially optimized and validated model might resolve the compilation issues I'm facing.

Thank you very much for your time and any information you can provide.