google-deepmind/mujoco_menagerie #201: Add ToddlerBot models

hshi74, 2025-08-29:
If you would like to see a new model added to the Menagerie collection, please provide us with the following information _before_ sending a pull request. (Ideally, before you start working on converting or tuning the model for MuJoCo.)

**URL for the existing model source (e.g. URDF)**
https://github.com/hshi74/toddlerbot/tree/main/toddlerbot/descriptions
https://github.com/hshi74/toddlerbot/tree/main/toddlerbot/descriptions/toddlerbot_2xc
We have multiple configurations, such as 2xc (default), 2xm (the same kinematics but with stronger Dynamixel motors), 2xc_gripper (with parallel grippers), 2xm_gripper. I'm unsure about whether I should add the default one, or add other configurations in some way as well.

**License terms of the upstream model**
MIT

**What do you think is required to convert and tune this model for MuJoCo?**
I believe I just need to format all the XML files according to contribute.md. Please let me know if you see something else.

**Are you proposing to help develop and maintain the new model?**
Yes.

kevinzakka, 2025-08-29:
Yes!

hshi74, 2025-09-25:
Any updates on this?